aredn/aredn #853: AFS Omission

Orv, 2023-05-28:
Using the AREDN Firmware Selector and typing sxt yields four Mikrotik devices with SXT in their device name;
MikroTik RouterBOARD SXT 2nD (SXT Lite2)
MikroTik RouterBOARD SXT 5HPnD (SXT 5 High Power)
MikroTik RouterBOARD SXT 5nD (SXT Lite5)
MikroTik SXTsq 5 ac (RBSXTsqG-5acD)

However, typing Mikrotik in the search field doesn't show these four devices.

aanon4, 2023-05-28:
I believe this issue has been filed before. If you type something general in the search box you will see the first N results, with a ... as the last entry indicating there are more. Please type something more specific.

Orv, 2023-05-28:
You're absolutely right - and that's completely non-obvious.

Orv, 2023-05-28:
However in this case, clicking on the ... yields nothing else.

aanon4, 2023-05-28:
An easier change to lobby for (I think) would be to include text in the description above the search box giving the searcher better information on how to find what they want. Not sure anyone is going to hack the JS for the actual search system.

It currently says:
```
Type the name or model of your device, then select a stable build or the nightly "snapshot" build (ie. 2050-781425a).
```

What would be better?

Orv, 2023-05-28:
Revising the verbiage might help, but the basic issue is still unchanged.  Those four devices aren't included when searching Mikrotik

aanon4, 2023-05-29:
We would need to find someone to take on a port of the OpenWRT firmware selector code so it will display a very long list of devices. They also need to maintain this with updates from OpenWRT.  That person isn't me but if someone else wants to do it, that'd be fantastic.

Failing that, we can add words to help people navigate the selector box successfully. This is easy to do today, and I think a good idea regardless of the first choice.

Orv, 2023-06-01:
That's not the issue in this case.    Those four devices aren't included when searching Mikrotik, regardless of how far you search into the returns yielded.

aanon4, 2023-06-03:
Not sure I understand this. There are about 27 Mikrotik devices available and if you just type "mikrotik" into the filter box you'll get the first 14. Unless you refine you search you're never going to see the remaining 13.