aredn/aredn #856: RB911G-2HPnD / mANTBox 2 12s listed as supported devices but images are not currently being made

linuxtuxguy, 2023-06-01:
I have noticed that for the past couple of releases (up to and including 3.23.4.0) there are no images being built for the Mikrotik RB911G-2HPnD and mANTBox 2 12s; instead on the supported devices link they just have a '-' listed for image name.

My club has several of these units in use, and they are stuck on version 3.22.12.0 due to lack of newer images.

The main status page shows the model string as being 'Mikrotik RouterBOARD RB911G-2HPnD'

The administration page shows the hardware type as '(ar71xx/mikrotik) mikrotik (rb-911g-2hpnd)'

If I choose the option on the Administration page to 'Refresh' the list of firmware it returns a 404 not found error (because the images are not being built and are not on the server):

```
Downloading firmware list from http://downloads.arednmesh.org/firmware...
Downloading
'http://downloads.arednmesh.org/firmware/firmware.rb-911g-2hpnd.list'
Connecting to 165.227.210.227:80
HTTP error 404
```

aanon4, 2023-06-01:
Because the changes in 3.23.4.0 were so extensive, we've only been able to provide images for devices we can test and unfortunately we dont have access to this hardware.  Is there anyway you could lend us a device so we can make the port and do the necessary testing?

linuxtuxguy, 2023-06-02:
> Because the changes in 3.23.4.0 were so extensive, we've only been able to provide images for devices we can test and unfortunately we dont have access to this hardware. Is there anyway you could lend us a device so we can make the port and do the necessary testing?

Yes, actually. I have one of these sitting new-in-box as a spare in case the ones in the field fail.

What would you need- I could potentially arrange for remote access to a linux vm connected to it, or if it's the only option I could ship it somewhere

linuxtuxguy, 2023-06-02:
I would also be fine applying any test images / providing feedback or diagnostics data on functionality if that's another option

eBebby, 2023-06-08:
First of all, thanks to the Dev Team for the great Job they do. As I have already written in the forum, I own the same antenna. Put this also gladly for tests online. Best regards from Switzerland.

aanon4, 2024-01-25:
https://github.com/aredn/aredn/pull/1083

closed 2024-03-05 by aanon4