aredn/aredn #859: Dual Band Devices Don't detect 5GHz non-mesh Wi-Fi devices Wi-Fi Scan in 5GHz Mesh Mode

k1ky, 2023-06-02:
With the new 5GHz Mesh feature in the recent nightlies, Dual-Band devices with the new 5GHz Mesh feature activated do not detect non-mesh regular Wi-Fi devices, showing only 5GHz Mesh devices, thus not giving a true picture of band usage in 20MHz bandwidth mode.  They do seem to work fine in 2.4GHz Mesh Mode @20Mhz bandwidth.  Single-Band 5 GHz Mesh MESH (non-AC) devices do show both MESH and Wi-Fi devices using 20MHz bandwidth.  Problem verified on  both HAP AC Lite and GL-INET AR750 (Creta) devices in 5GHz Mesh Mode.  The 2.4 GHz Mesh mode does show both MESH and regular Wi-Fi devices on these models.

aanon4, 2023-06-02:
This is what I get running nightly on my HAP with the mesh radio set to 5GHZ/20MHz:

[image]

The only observation I can make is that I had to refresh the screen a few times to get this - it certainly didnt come up every time. But I've found that to be the case with wifi scanners in general so this isn't something new.

k1ky, 2023-06-02:
I put it on "auto" still no joy on AR750 Creta. I have both radios enabled - 2.4GHz is operating as Wi-Fi AP and 5GHz Mesh..  Is that how you are operating?  Continuing to evaluate with Nightly 0602.

k1ky, 2023-06-02:
I changed the MESH Channel to an unoccupied channel 36 and got a big Wi-Fi station list on auto for 1 second and then it went back to showing only the host node and no other stations. HAP AC Lite. Nightly 20230602

aanon4, 2023-06-02:
I think we have to conclude this is a limitation of some of the hardware. If you type the scan command by hand:

```iw wlan0 scan```

Then you'll see a (usually) short list. There's nothing much I can do about this if the hardware doesnt report more.

k1ky, 2023-06-02:
Verified that 5GHz Wi-Fi scan shows all devices when 2.4GHz radio is disabled and node is connected to a computer via Ethernet.

k1ky, 2023-06-03:
I updated the description and text in the opening statement based on further data collected.  Seems that the 5GHz Wi-Fi scan doesn't show non-Mesh devices only when "Connected" to another 5GHz Mesh node. If there is no Mesh connection, then the 5GHz WI-Fi scan will show all devices in the band, Wi-Fi or Mesh.

closed 2023-07-19 by ab7pa

k1ky, 2024-07-06:
The Mikrotik HAP3 is able to display ALL 5GHz stations - Mesh and Non-Mesh. 
[image]