aredn/aredn #869: Ubiquiti RM5 XW: 3.23.4.0: holding reset for 5 sec resets node to just-flashed state

Rom3oDelta7, 2023-06-12:
The documentation says that holding the reset button down for 5 sec will reset the password and DHCP state to default. However, doing so instead resets the node back to just-flashed state. This occurs whether the node reset button or the power brick reset button is pressed. It is supposed to take 15 sec to trigger a full node reset. I have tried the timing 4 times while watching a stopwatch, and holding for 5+ (but less than 6) seconds incorrectly does a full reset.

[supportdata-K7YCA-RM5-XW-SPARE-202306120101.gz](https://github.com/aredn/aredn/files/11716435/supportdata-K7YCA-RM5-XW-SPARE-202306120101.gz)

aanon4, 2023-06-12:
I'll look to see if the latest OpenWRT has changed how this works, but the configuration (/etc/config/system) today says:

Press for 3-7 second - reset password
Press for 12-20 seconds - return to firstboot state

aanon4, 2023-06-12:
Looks like our old handlers never get to be called because procd take over first. Will look at fixing this.

aanon4, 2023-06-12:
Could you try this for me to see if it resolve the problem:

Log into you node
Remove the file /etc/rc.button/reset
Reboot

When the node comes back up, see if the expected reset behavior is restore.

Rom3oDelta7, 2023-06-14:
I'll try that tomorrow and report back. Thank you for looking into this.

73, Rob

Rom3oDelta7, 2023-06-15:
I ssh'd into my test Loco-M2-XW node running 3.23.4.0 and renamed the /etc/rc.button/reset file and rebooted.

Reboot was normal, so I logged in with my current root password.

Held reset for 6 seconds. The node did not reboot, but the password _was_ reset to the default. I think that _not_ rebooting is the proper behavior, but it's been a while since I had to do this and I don't remember. Regardless, the desired reset behavior was observed.

73, Rob

aanon4, 2023-06-23:
https://github.com/aredn/aredn/pull/879

closed 2023-07-19 by ab7pa