aredn/aredn #880: Multiple firmware device match failed

kc2cbd, 2023-06-24:
I've been getting "firmware device match failed" on the following devices.
Mikrotik Basebox mikrotik-912uag-5hpnd
Mikrotik QRT mikrotik-911g-5hpnd-qrt
TP-Link CPE510 v3 tplink_cpe510-v3
This has been over the past several version of nightlies and I don't remember how far back it started.
Thanks

aanon4, 2023-06-24:
Can you provide the following pieces of information for the Mikrotik Basebox (I have one of these somewhere):
1. What specific firmware version is it running now.
2. What specific firmware version it fail to upgrade to.
3. A screen shot showing all the information on the admin screen after the upgrade fails.
4. A support tool data dump immediately after the upgrade fails.

Thanks

kc2cbd, 2023-06-25:
Ok can't get the TP-Link right now but the Basebox's and the QRT seem to work correctly tonight.

kc2cbd, 2023-06-25:
Has 20230623 and trying to use 20230624
[image]
[supportdata-KQ4CIN-CPE510-102-CKV-25TWR-186-202306250344.gz](https://github.com/aredn/aredn/files/11859206/supportdata-KQ4CIN-CPE510-102-CKV-25TWR-186-202306250344.gz)

aanon4, 2023-06-25:
Well this is weird. Nothing significant changed between the version you're running and the version you were trying to upgrade to.

The only time I've seen this before is when the upgrade file is somehow corrupted while being uploaded. Are you upgrading using the "Upload Firmware" option, the "Download Firmware" or the "Load Local Firmware" option?

kc2cbd, 2023-06-25:
I've tried with the "Download Firmware" option first but that usually fails with "firmware file checksum failed" then I use "Upload Firmware" option, which is what happened with this one.

closed 2023-06-26 by kc2cbd