aredn/aredn #884: Ubiquiti Litebeam 5AC: downgrade to Firmware version 8.7.0 rejected

HB9XCL, 2023-06-26:
Following the instructions, a Ubiquiti Litebeam 5AC WA needs to be downgraded to firmware version 8.7.0 before uploading the AREDN firmware. This however doesn't work in case of a firmware version > 8.7.4 is installed on the device - the device rejects the upload, error message: **_"Incorrect firmware version. Please upload v8.7.4 or newer"_**.

I've tried the following: 

- Install v8.7.4 and install v8.7.0 from here -> still same error message 
- TFTP the v8.7.0 -> leads to a boot loop, the device doesn't start up anymore.
- Install v8.7.4Beta -> radio was bricked

There won't be a many (any?) of these devices on the market having firmware 8.7.3 or lower installed, so more and more users will run into this problem. 
Thanks for checking.
Kurt HB9XCL

BKasin, 2023-09-02:
If the issue still is happening for you, apparently firmware version WA.v8.7.4-beta.44660.210219.1654.bin allows a downgrade.

https://dl.ubnt.com/11AC-fw-internal/v8.7.4/WA.v8.7.4-beta.44660.210219.1654.bin

aanon4, 2023-10-05:
Can someone point me at the store the problematic hardware came from? I tried buying a new LiteBeam from Amazon but it came with 8.7.1 on it and will both upgrade to 8.7.11 and downgrade to 8.7.0 without a problem. Looks like I need access to the specific hardware if I'm to solve this problem.

aanon4, 2023-11-15:
Fixed in nightly.

closed 2023-11-15 by aanon4