aredn/aredn #892: Custom Routing does not survive Save Changes/Reboot sequence

n2mh, 2023-07-07:
Nightly 20230704-eaa5d2f 
In /etc/config/network there is a section where custom static routes can be entered.
         
### Static routes              
#                                    
# Add extra static routes here.         
# For details see: https://openwrt.org/docs/guide-user/network/routing/routes_configuration

I have added 2 stanzas to this section as follows

config  route 'ampr_low'                                                                   
        option interface 'wan'       
        option target '44.0.0.0'       
        option netmask '255.128.0.0'    
        option gateway '44.64.24.128'
                                                                                           
config  route 'ampr_high'             
        option interface 'wan'         
        option target '44.128.0.0'      
        option netmask '255.192.0.0' 
        option gateway '44.64.24.128' 

After saving them and issuing a reboot command on the command line, these routes *do* show up in ip route

44.0.0.0/9 via 44.64.24.128 dev br-wan 
44.128.0.0/10 via 44.64.24.128 dev br-wan 

However, if I go to the Node's Setup page and do a [Save Changes] followed by a [Reboot] sequence, those custom routes are missing when the node comes back to life.

Since this section of /etc/config/network specifically has a section for adding additional static routes, along with an example, I would have to assume that this custom work would survive the Save Changes/Reboot sequence.

Unless I'm blind, Read-The-Docs is silent on this subject. (Maybe my sight *is* going? :)

73, Mark, N2MH

n2mh, 2023-07-07:
This is on a hAP Lite.

aanon4, 2023-07-08:
Because we blow away the /etc/config/network file when you save changes or reinstall, we provide /etc/aredn_include/static_routes as a place to put your extra routes. I guess we dont document this yet.

ab7pa, 2023-07-19:
This feature is documented here: https://docs.arednmesh.org/en/latest/arednServicesGuide/network_tools.html#manage-extra-static-routes

closed 2023-07-19 by ab7pa