aredn/aredn #893: mesh RF address clash

nc8q, 2023-07-07:
I have a Mikrotik hAP and a GL-iNet AR750 that have a 'mesh RF address' clash.
These devices do not play well when on the same network.
How can I get them both to work on the same network at the same time?
hap image http://gelm.net/AREDN/hap2-32340.png
ar750 image http://gelm.net/AREDN/ar750-32340.png

73, Chuck

aanon4, 2023-07-08:
In theory you can just enter a new 10.X.X.X address on the main setup page under Mesh. Never tried it though so would be interested in how well it works.

nc8q, 2023-07-10:
I reinstalled 3.23.4.0 on my newer hAP using TFTP and got the same Mesh IP address.
I think this is a 1 in 256*256*256 chance of happening. ?
This duplication is from different manufacturers!
A few attempts to enter a different IP address was rejected but I found one that was accepted.
I will try out this address.
I will close this issue.
Thanks, Chuck

closed 2023-07-10 by nc8q