aredn/aredn #896: My new SXTsq-5ac device will not link at > 54 TxMbps

nc8q, 2023-07-10:
I have a SXTsq-5ac and a LHGg-5ac-xl and neither
links with each other or my other 5[HP]nd devices 
at > 54 TxMbps.
[supportdata-NC8Q-SXTsq-5ac-202307101024.gz](https://github.com/aredn/aredn/files/12000838/supportdata-NC8Q-SXTsq-5ac-202307101024.gz)

nc8q, 2023-07-19:
Added 2 files: output from 
iw dev wlan0 station dump
on my linked SXTsq-5ac and LHG-5nD.
http://gelm.net/AREDN/896/

ae6xe, 2023-07-19:
unable to see supportdata.  see attachment of error. 
[image]

nc8q, 2023-07-20:
On 7/19/23 14:13, Joe AE6XE wrote:
>
> Would also like to capture the output of "iw dev wlan0 scan" on both 
> devices. Looking for the beacon for each device (these stations):
> 64:d1:54:9c:85:af
> 74:4d:28:92:be:26
>

Hi, Joe:

Well, I don't have the LHGG-5ac device on hand, but I do have a
LHG-5nD and a SXTsq-5ac device linked.
I ran 'iw dev wlan0 scan > /tmp/ac-iw-dev-wlan0-scan.txt' and
'iw dev wlan0 scan > /tmp/n-iw-dev-wlan0-scan.txt' and transferred
those files to http://gelm.net/AREDN/896/

I hope you can find what you asked for among the files there.

Chuck

ae6xe, 2023-07-25:
I have confirmed that the hap ac2 device is limiting to 802.11b rates -- up to 54Mbps -- 5GHz AC ath10k interface.   This is the state before the device is connected to another device (no link).   The node from the beginning is only advertising in the beacon to be compatible with 802.11b rates up to 54Mpbs.  See attached pic.   I suspect this is related to configuring and/or getting the ath10k driver working on 802.11n/ac IBSS/adhoc mode with the various chipsets on all these devices.   

[image]

I have seen reports of AC devices connected and exceeding these rates.   If we can find a pattern, where this is working and not working, it may lead us to the root cause.  

To find or confirm if a node is becon'ing only compatibility with 802.11b, here is a way to capture.   If others can post the results on various devices, we might be able to find a pattern.     It does not appear to be related to combinations of 'n' and 'ac' devices.   I would not expect this hap ac2 device to do an RF link on 5Ghz (802.11ac hardware) to any device 'n' or 'ac' beyond 802.11b rates.

1) install the tcpdump optional package
2) turn on mesh RF (on an AC interface) and set channel, SSID, or channel width such that the node will not link to another node.  (avoid there being other traffic on the channel, so we don't have to comb through  the data to find beacons.)
3) on the node, run this command to capture the packets on the 'monitor' interface.  On nodes with multiple interfaces, have to get the right one (wlan0-1 or wlan1-1):    "tcpdump -i wlan0-1 -w /tmp/capture.pcap -c 30"
4) copy the data file  down to your computer and upload here for review.  Example from a Windows power shell command:   "scp -P 2222 root@localnode.local.mesh:/tmp/capture.pcap ./"   (be sure to rename the file to something more descriptive when uploading.)
5) to inspect the data and see the beacons, install "wireshark", and open up this packet capture (.pcap) file. 

Joe AE6XE

nc8q, 2023-07-26:
On 7/25/23 17:49, Joe AE6XE wrote:
> # on the node, run this command to capture the packets on the 'monitor' 
> interface. On nodes with multiple interfaces, have to get the right 
> one (wlan0-1 or wlan1-1): "tcpdump -i wlan0-1 -w /tmp/capture.pcap -c 30"
> # copy the data file down to your computer and upload here for review.

nc8q-sxtsq-5ac.pcap attached.

ae6xe, 2023-07-27:
Not finding the uploaded .pcap file?

aanon4, 2025-04-21:
Made a change yesterday and I'd be interested if it affected this problem or not.

nc8q, 2025-04-22:
My two Mikrotik SXT-5ac devices now link at close to 144 Mbps.
:-)
Thanks.

closed 2025-04-22 by nc8q