aredn/aredn #898: Bullet AC - Unsupported (2WA) Is there support for this?

ASULTDGIT, 2023-07-12:
node: NOCALL-96-201-188
model: 0xe2c5

!!!! UNSUPPORTED DEVICE !!!!
boardid: 0xe2c5

br-lan E0:63:DA:61:C9:BC
eth0   E0:63:DA:61:C9:BC
eth0.1 E0:63:DA:61:C9:BC
eth0.2 E0:63:DA:61:C9:BC
tunl0  00-00-00-00-00-00-00-44-00-00-00-00-00-00-00-00
wlan0  E0:63:DA:60:C9:BC
wlan1  E2:63:DA:60:C9:BC

/proc/cpuinfo
system type		: Atheros AR9342 rev 3
machine			: Ubiquiti Bullet AC (2WA)
processor		: 0
cpu model		: MIPS 74Kc V4.12
BogoMIPS		: 266.64
wait instruction	: yes
microsecond timers	: yes
tlb_entries		: 32
extra interrupt vector	: yes
hardware watchpoint	: yes, count: 4, address/irw mask: [0x0ffc, 0x0ffc, 0x0ffb, 0x0ffb]
isa			: mips1 mips2 mips32r1 mips32r2
ASEs implemented	: mips16 dsp dsp2
Options implemented	: tlb 4kex 4k_cache prefetch mcheck ejtag llsc dc_aliases perf_cntr_intr_bit perf
shadow register sets	: 1
kscratch registers	: 0
package			: 0
core			: 0
VCED exceptions		: not available
VCEI exceptions		: not available

nvram
hsmmmesh.settings=settings
hsmmmesh.settings.wifimac='e0:63:da:60:c9:bc'
hsmmmesh.settings.mac2='96.201.188'
hsmmmesh.settings.node='NOCALL-96-201-188'
hsmmmesh.settings.dtdmac='97.201.188'

aanon4, 2023-07-12:
Is this a 2.4GHz device? I've had no luck getting 2.4GHz Ubiquiti AC devices working so far.

ASULTDGIT, 2023-07-12:
the bullet ac is dual band :)

aanon4, 2023-07-12:
I guess the next questions is, what works?

ASULTDGIT, 2023-07-13:
i tried the most recent release under the software and, it doesnt even allow point in which to get to set up, just hits 504 bad gateway, tried the nightly, was able to get to set up page but same thing nothing worked.

aanon4, 2023-07-13:
There are dozens of aredn firmware binaries to choose from, so could I ask which one you're trying? And yes, just go with a nightly as the release *really* doesnt like unsupported hardware.

ASULTDGIT, 2023-07-14:
I chose the ubqt bullet ac 2023 stable release 
3.23.4.0 (r20028-43d71ad93e)

aanon4, 2023-07-14:
We dont list that as supported as no one had tested it.
Could you elaborate on what does and doesnt work. You say "nothing worked" but ... really nothing? No LAN, DtD, DHCP .. nothing?

ASULTDGIT, 2023-07-15:
Yea, with that release I stated, when I flashed it, and you get to the initial aredn page, and you click the setup button to get to the admin page where you set all those details, dhcp etc. you are presented with 504 bad gateway. That is it. Unable to proceed past it, so really nothing worked.

aanon4, 2023-07-15:
Yes, a release wont get you far as it's very unforgiving of unknown devices. But please try a nightly which should allow you to get further.

ASULTDGIT, 2023-08-02:
Update:

Using nightly 20230709:

Things that work:
- Lan DHCP
- setup page 
- able to put in any config

Things that do not work:
- everything RF
- unable to connect to other nodes/pick them up
- unable to connect to wifi wan client
- unable to pick up anything on wifi scan
- SNR sits at a constant -95dB at any distance between the nodes
- issues persist on both freq bands 
5/2.4

aanon4, 2023-09-06:
After some discussion, we've decided not to support new SIMO devices and only support MIMO going forward.

closed 2023-09-06 by aanon4