aredn/aredn #901: Suggestion - Switch new advanced access settings default to ON

km6lslf, 2023-07-18:
A couple times I've updated my nodes to the current nightly, I wasn't able to access it like I was before, and needed to go through a couple extra steps to get back into it before I could set the new access settings to ON.

Enable web access to the node from the WAN interface
aredn.@wan[0].web_access
Enable SSH access to the node from the WAN interface
aredn.@wan[0].ssh_access

I like having these options, but I'm concerned that new node operators may assume the AREDN firmware is broken if they flash the firmware and can't access the node. Resetting it to firstboot status wouldn't even restore access for them - they'd need to figure it out on their own or go to forums, making for a worse user experience. 

I strongly suggest switching the default setting for these two settings to ON.

aanon4, 2023-07-18:
Happy to see discussion on this. From my perspective we added these options to make things more secure, and leaving them on by default doesnt achieve that goal. So I prefer off by default.

km6lslf, 2023-07-18:
Enhanced security is a fine goal and has my full support, but not at the cost of UX, especially for new or inexperienced users. 

What do you think or recommending they be off, but having them on by default to prevent users from having issues administering their nodes?

aanon4, 2023-07-19:
So after some discussion this evening with the devs it was agreed that that the firewall should be open by default as you suggest here. Look for an incoming fix later today.

aanon4, 2023-07-19:
https://github.com/aredn/aredn/pull/903

closed 2023-07-19 by ab7pa