aredn/aredn #902: Feature request: Terminal command to set Allow other MESH nodes to use my WAN setting to ON

km6lslf, 2023-07-18:
In a recent discussion with a fellow RACES member about potential operational scenarios, there were multiple situations where quickly turning on multiple nodes Allow other MESH nodes to use my WAN setting to on would be very helpful. 

If this can be done over SSH, that would allow the command to be sent to all desired nodes by an authorized user using a script rather than logging into each node individually and tweaking the setting. Can this be done currently? If not, could the feature be added?

aanon4, 2023-07-18:
I would recommend looking at Ansible which is a tool design for setting up configurations on multiple systems remotely in a very flexible way. I use it myself for updating and running tests on my two dozen test nodes.

Actually modifying this particular setting is relatively easy. On the Advanced Configuration page on each setting you'll see some small text reading, for example, aredn.@wan[0].olsrd_gw. This is a property in the UCI system (see OpenWRT for more information on that) which can be modified from the command line.

km6lslf, 2023-07-18:
Thanks, I was considering Ansible for this. 

I ran the uci set aredn.@wan[0].olsrd_gw='1' command, then uci commit

When set that way it didn't seem to actually provide internet access to a node 1 hop away. I noticed it wasn't prompting me to reboot, but I rebooted anyway hoping it may help.  The setting persisted in the ON status, but the mesh connected node still doesn't seem to get internet access from it. Any ideas?

aanon4, 2023-07-18:
From memory (so please check by looking at the code in the advanced configuration page) it goes something like this:
```
uci set aredn.@wan[0].olsrd_gw='1'
uci -c /etc/config.mesh set aredn.@wan[0].olsrd_gw='1'
/usr/loca/bin/node-setup -a mesh
reboot
```

km6lslf, 2023-07-18:
Thanks! For anyone else following along, there does not appear to be usr directory when I used ls. I used find and found the file at /rom/usr/local/bin/node-setup

Still no luck on getting internet passed to a node 1 hop away... tested with 2 different nodes now.

aanon4, 2023-07-19:
Bit odd that you didn't find a /usr directory. Nothing would work if that was missing.

Also, I didn't get it quite right.
```
uci set aredn.@wan[0].olsrd_gw='1'
uci -c /etc/config.mesh set aredn.@wan[0].olsrd_gw='1'
uci  commit aredn
uci -c /etc/config.mesh commit aredn
/usr/local/bin/node-setup -a mesh
reboot
```

closed 2023-07-19 by ab7pa

km6lslf, 2023-07-19:
Did that work for you when you tried it? I had to add "rom" before the /usr/... but aside from that the commands look like they executed OK, but when I checked the node it didn't have the setting turned on. I turned it on in the advanced configuration page, but still couldn't get internet on a mesh-connected Nanostation on my roof... 

Could this be a bug on haps?