aredn/aredn #905: support tool: trailing garbage ignored

nc8q, 2023-07-20:
The 'download support tool' in nightly 202307[05 and 19] ungzip returns:

gelmce@nc8q-acer:~/Downloads$ cd tmp
gelmce@nc8q-acer:~/Downloads/tmp$ gunzip ../supportdata-NC8Q-SXTsq-5ac-202307112019.gz 

gzip: ../supportdata-NC8Q-SXTsq-5ac-202307112019.gz: decompression OK, trailing garbage ignored
gelmce@nc8q-acer:~/Downloads/tmp$ gunzip ../supportdata-NC8Q-SXTsq-5ac-202307112104.gz 

gzip: ../supportdata-NC8Q-SXTsq-5ac-202307112104.gz: decompression OK, trailing garbage ignored
gelmce@nc8q-acer:~/Downloads/tmp$

nc8q, 2023-07-20:
http://gelm.net/AREDN/896/

aanon4, 2023-09-16:
As far as I can tell, the "trailing garbage" is unimportant. If this is just annoying but not stopping us getting the support data, I'll leave unfixed for now. Please reopen if this is not the case.

closed 2023-11-15 by aanon4