aredn/aredn #906: HAP AC3 loss of WIFI WAN on upgrade to nightly

kd2hkb, 2023-07-20:
When I upgraded a HAP AC3 to the 20230704 nightly... I lost the "wifi client" option as well as the AP configuration options on the "Basic Setup" tab. I have since upgraded to the 20230719 nightly and those options did not come back. I have also reverted it back to the latest stable & the options re-appeared, but when I again upgraded to the 20230719 nightly... the options disappeared again.
[image]

EffectiveAgileDev, 2023-07-20:
I had the same problem on an AC lite.  Asking locally about the problem, I got a question "do you have RF on?"  I did not, so I turned it on and rebooted.  When the device came back up all of the optional items were there and RF was OFF!

aanon4, 2023-07-20:
This looks suspiciously like some kind of rendering error. Unfortunately the currently nightly renders just fine on my ac3. Could someone with this problem:
1) Install the current nightly.
2) Try to load this setup page.
3) Immediately after, generate and send the support data.
Thanks

kd2hkb, 2023-07-21:
Working on a different node than originally posted about...

Pre-Upgrade support data (for info)

[supportdata-NC2C-T2-202301051236.gz](https://github.com/aredn/aredn/files/12122690/supportdata-NC2C-T2-202301051236.gz)

Post-Upgrade support data

[supportdata-NC2C-T2-202307211035.gz](https://github.com/aredn/aredn/files/12122719/supportdata-NC2C-T2-202307211035.gz)

WHAT THE... Sorry... that one went as I would have expected!

[image]

kd2hkb, 2023-07-21:
While I am now thinking that this is more of just a random issue vs a real bug... here is the support data from the node which does not load correctly.

[supportdata-NC2C-T1-202307211159.gz](https://github.com/aredn/aredn/files/12123295/supportdata-NC2C-T1-202307211159.gz)

This node is "In Service" for an event this weekend, but after the event... I think I'll try to reflash the node just to see what it does then.

aanon4, 2023-07-21:
Oh it looks like a real bug. There's an error while it's generating this page - you can see it in the data you sent (thanks). I'll look into the "why" later today.
Thanks.

aanon4, 2023-07-21:
Any possibility you could send the output from running these two commands on the node?
```
iwinfo wlan0 freqlist
iwinfo wlan1 freqlist
```

kd2hkb, 2023-07-21:
`login as: root
root@nc2c-t1.local.mesh's password:

BusyBox v1.35.0 (2023-04-27 20:28:15 UTC) built-in shell (ash)

              _____  ______ _____  _   _
        /\   |  __ \|  ____|  __ \| \ | |TM
       /  \  | |__) | |__  | |  | |  \| |
      / /\ \ |  _  /|  __| | |  | | . ` |
     / ____ \| | \ \| |____| |__| | |\  |
    /_/    \_\_|  \_\______|_____/|_| \_|
    AMATEUR  RADIO EMERGENCY DATA NETWORK
-----------------------------------------------
 1) Research AREDN and choose a supported device
 2) Download and install AREDN firmware
 3) Deploy and enjoy the mesh
-----------------------------------------------
  20230719-ef17d9b, r20134-5f15225c1e
 ----------------------------------------------
root@NC2C-T1:~# iwinfo wlan0 freqlist
No such wireless device: wlan0
root@NC2C-T1:~# iwinfo wlan1 freqlist
* 5.180 GHz (Channel 36)
  5.200 GHz (Channel 40)
  5.220 GHz (Channel 44)
  5.240 GHz (Channel 48)
  5.260 GHz (Channel 52)
  5.280 GHz (Channel 56)
  5.300 GHz (Channel 60)
  5.320 GHz (Channel 64)
  5.500 GHz (Channel 100)
  5.520 GHz (Channel 104)
  5.540 GHz (Channel 108)
  5.560 GHz (Channel 112)
  5.580 GHz (Channel 116)
  5.600 GHz (Channel 120)
  5.620 GHz (Channel 124)
  5.640 GHz (Channel 128)
  5.655 GHz (Channel 131)
  5.660 GHz (Channel 132)
  5.665 GHz (Channel 133)
  5.670 GHz (Channel 134)
  5.675 GHz (Channel 135)
  5.680 GHz (Channel 136)
  5.685 GHz (Channel 137)
  5.690 GHz (Channel 138)
  5.695 GHz (Channel 139)
  5.700 GHz (Channel 140)
  5.705 GHz (Channel 141)
  5.710 GHz (Channel 142)
  5.715 GHz (Channel 143)
  5.720 GHz (Channel 144)
  5.725 GHz (Channel 145)
  5.730 GHz (Channel 146)
  5.735 GHz (Channel 147)
  5.740 GHz (Channel 148)
  5.745 GHz (Channel 149)
  5.750 GHz (Channel 150)
  5.755 GHz (Channel 151)
  5.760 GHz (Channel 152)
  5.765 GHz (Channel 153)
  5.770 GHz (Channel 154)
  5.775 GHz (Channel 155)
  5.780 GHz (Channel 156)
  5.785 GHz (Channel 157)
  5.790 GHz (Channel 158)
  5.795 GHz (Channel 159)
  5.800 GHz (Channel 160)
  5.805 GHz (Channel 161)
  5.810 GHz (Channel 162)
  5.815 GHz (Channel 163)
  5.820 GHz (Channel 164)
  5.825 GHz (Channel 165)
  5.830 GHz (Channel 166)
  5.835 GHz (Channel 167)
  5.840 GHz (Channel 168)
  5.845 GHz (Channel 169)
  5.850 GHz (Channel 170)
  5.855 GHz (Channel 171)
  5.860 GHz (Channel 172)
  5.865 GHz (Channel 173)
  5.870 GHz (Channel 174)
  5.875 GHz (Channel 175)
  5.880 GHz (Channel 176)
  5.885 GHz (Channel 177)
  5.890 GHz (Channel 178)
  5.895 GHz (Channel 179)
  5.900 GHz (Channel 180)
  5.905 GHz (Channel 181)
  5.910 GHz (Channel 182)
  5.915 GHz (Channel 183)
  5.920 GHz (Channel 184)
root@NC2C-T1:~#
`

aanon4, 2023-07-21:
Thanks.
So while I can protect against this error (and will), the bigger question is - why do you not have wlan0 ? I'm going to have to try to reproduce it on my ac3 (so far no luck).

kd2hkb, 2023-07-21:
Not a clue. When I upgraded I had Mesh off, WIFI LAN on & WIFI WAN on. If that helps... I do not know. 

I was just doing some experimenting, but now I have a bigger problem... the node dropped from the network completely. And it's deployed for an event this weekend. Guess I better stop by that site on my way by tonight! :D

One experiment I did was revert back to the last stable. When I did... the entire page showed, but running the commands mentioned about yielded the same results. No WLAN0... only WLAN1

aanon4, 2023-07-21:
Thanks for the info.
The last stable release just assumed the devices where there, while the new firmware actually looks for them. That mostly why the page renders correctly on the release (it doesnt actually check). But that wlan0 is missing in both cases is a little worrying. Have you tried power cycling (rather than just a reboot)?

aanon4, 2023-07-21:
https://github.com/aredn/aredn/pull/907

This should at least make the UI tolerate this problem (if you have a chance to test it, that'd be good) though it does nothing for the root cause I'm afraid.

kd2hkb, 2023-07-25:
Well... good news. Re-flashing from scratch fixed things.

root@NC2C-T1:~# iwinfo wlan0 freqlist
  2.387 GHz (Channel -4)
  2.392 GHz (Channel -3)
  2.397 GHz (Channel -2)
  2.402 GHz (Channel -1)
  2.412 GHz (Channel 1)
  2.417 GHz (Channel 2)
  2.422 GHz (Channel 3)
  2.427 GHz (Channel 4)
  2.432 GHz (Channel 5)
  2.437 GHz (Channel 6)
  2.442 GHz (Channel 7)
  2.447 GHz (Channel 8)
  2.452 GHz (Channel 9)
  2.457 GHz (Channel 10)
  2.462 GHz (Channel 11)
  2.467 GHz (Channel 12)
  2.472 GHz (Channel 13)
  2.484 GHz (Channel 14)
root@NC2C-T1:~# iwinfo wlan1 freqlist
* 5.180 GHz (Channel 36)
  5.200 GHz (Channel 40)
  5.220 GHz (Channel 44)
  5.240 GHz (Channel 48)
  5.260 GHz (Channel 52)
  5.280 GHz (Channel 56)
  5.300 GHz (Channel 60)
  5.320 GHz (Channel 64)
  5.500 GHz (Channel 100)
  5.520 GHz (Channel 104)
  5.540 GHz (Channel 108)
  5.560 GHz (Channel 112)
  5.580 GHz (Channel 116)
  5.600 GHz (Channel 120)
  5.620 GHz (Channel 124)
  5.640 GHz (Channel 128)
  5.655 GHz (Channel 131)
  5.660 GHz (Channel 132)
  5.665 GHz (Channel 133)
  5.670 GHz (Channel 134)
  5.675 GHz (Channel 135)
  5.680 GHz (Channel 136)
  5.685 GHz (Channel 137)
  5.690 GHz (Channel 138)
  5.695 GHz (Channel 139)
  5.700 GHz (Channel 140)
  5.705 GHz (Channel 141)
  5.710 GHz (Channel 142)
  5.715 GHz (Channel 143)
  5.720 GHz (Channel 144)
  5.725 GHz (Channel 145)
  5.730 GHz (Channel 146)
  5.735 GHz (Channel 147)
  5.740 GHz (Channel 148)
  5.745 GHz (Channel 149)
  5.750 GHz (Channel 150)
  5.755 GHz (Channel 151)
  5.760 GHz (Channel 152)
  5.765 GHz (Channel 153)
  5.770 GHz (Channel 154)
  5.775 GHz (Channel 155)
  5.780 GHz (Channel 156)
  5.785 GHz (Channel 157)
  5.790 GHz (Channel 158)
  5.795 GHz (Channel 159)
  5.800 GHz (Channel 160)
  5.805 GHz (Channel 161)
  5.810 GHz (Channel 162)
  5.815 GHz (Channel 163)
  5.820 GHz (Channel 164)
  5.825 GHz (Channel 165)
  5.830 GHz (Channel 166)
  5.835 GHz (Channel 167)
  5.840 GHz (Channel 168)
  5.845 GHz (Channel 169)
  5.850 GHz (Channel 170)
  5.855 GHz (Channel 171)
  5.860 GHz (Channel 172)
  5.865 GHz (Channel 173)
  5.870 GHz (Channel 174)
  5.875 GHz (Channel 175)
  5.880 GHz (Channel 176)
  5.885 GHz (Channel 177)
  5.890 GHz (Channel 178)
  5.895 GHz (Channel 179)
  5.900 GHz (Channel 180)
  5.905 GHz (Channel 181)
  5.910 GHz (Channel 182)
  5.915 GHz (Channel 183)
  5.920 GHz (Channel 184)
root@NC2C-T1:~#

kd2hkb, 2023-08-17:
I have no suggestions. I have access to 3 of these devices. Two of them did the same exact thing as I showed above. Reflashing using the TFTP method and the most recent nightly build seemed to fix the issue.

k1ky, 2023-10-02:
We encountered a "new" version of the HAP3 board that may have been similar to this.  Current Nightlies have resolved this issue on these units.  Possibly this issue as well?  There is still an issue with HAP3 nightly upgrades where the MESH Radio doesn't come back online and requires a Wi-Fi scan to bring it alive.  This may need to be another "issue" if this behavior persists.

kc7dmf, 2024-01-29:
I am using 
firmware version 3.23.12.0
model: MikroTik hAP ac3
This isn't even the "nightly, but same results... 0.5 up, 178 down out of expected 20 up 600 down
Ethernet, I get 24 up and 178 down. 

I wonder if the ac2 is a better choice?

The original hAP caps at 100Mbps, not much of a point in Gig ports and speed barely over 100Mbps.

aanon4, 2024-03-29:
Any updates on this with the current nightly builds?