aredn/aredn #910: Ubiquiti nanostation loco m5

n2jwc, 2023-08-05:
Had one of my nanostation loco m5 xw go offline, it is attached to a hAP ac lite which is doing POE pass-thru. It has been offline for 24.4 hours and I had to power cycle it via the hAP (or unplug it). The same setup on the 'other side' of the link.

HOME SIDE ----->>>> mesh REMOTE SIDE
Internet ----- [hAP ac lite]----POE----[NANO]  ))))))) mesh ((((((( [NANO]----POE----[hAP ac lite] --- Camera

All this is a bench setup on mains power as I test and ensure all works as I hoped it would. I have downloaded the support files for the remote side nano and hAP.  Not sure if logs survive a reboot for the nano but who knows.

[supportdata-N2JWC-SHOP-202301032251.gz](https://github.com/aredn/aredn/files/12267148/supportdata-N2JWC-SHOP-202301032251.gz)

[supportdata-N2JWC-SHOP-HAP-202308051309.gz](https://github.com/aredn/aredn/files/12267149/supportdata-N2JWC-SHOP-HAP-202308051309.gz)

aanon4, 2023-09-16:
Is there a way you could power the nanostation loco m5 separately to see if this is a random power issue?

n2jwc, 2023-09-16:
Hi. I did take it off the POE from the MikroTik and used the native POE injector. Still failed after some time. 

Swapped it with the “opposite” side node and still failed after time. Only this time with Smokeping running I could see the RF was up but the LAN was down.