aredn/aredn #912: RB951Ui-2ND Firmware

do1pps, 2023-08-15:
Hello AREDN Team,

by a lucky coincidence I got a lot of Mikrotik RB951Ui-2ND at a very reasonable price.

Unfortunately there is no firmware for this type of router available on arednmesh.org website.

Could you publish a firmware for these routerboards?

Thanks for your efforts around the topic AREDN

With kind regards

Patrick - DO1PPS

CocopeliTR, 2023-09-03:
Dear amateur radio friend;

If you read this link, you can use that model device for routing to AREDN. 

https://www.arednmesh.org/content/mikrotik-tutorial

TA1DE

aanon4, 2023-11-15:
This device is a bit old, with only 100Mb ports. I dont think we can add it to the official support list.

closed 2023-11-15 by aanon4

InspiringCode, 2025-02-09:
Sorry to reopen this, but is the supported now or not? In the changelog for [3.23.12.0 (2023-12-04)](https://downloads.arednmesh.org/snapshots/CHANGELOG.md) it says "RB951Ui-2ND Firmware #912" but it isn't listed on the supported devices list. Does anyone know what the current status is?

I find this device interesting, because it is one of the cheapest 128MB models with 2.4ghz support.

aanon4, 2025-02-09:
That just the change log generator being problematic. This is not supported.