aredn/aredn #919: [Solved] MikroTik hAP ac3 WLAN interface disappear / new ac3 comes in RouterOS v7

SCWhite, 2023-08-30:
### TL;DR
If you have new MikroTik hAP AC3
modify openwrt and build AREDN firmware from scratch

### So here is the story:
I got 3 of them, 1 old, 2 new.
the old one come with routerOS v6 (MAC DC:2C:6E:XX:XX:XX)
newer ones have routerOS v7 (MAC 48:A9:8A:XX:XX:XX)

I follow the document, and the old one works perfectly.
but no mater how I try to flash those new ac3, my wireless interface doesn't function at all.
(it seems to be broken on every nightly build / 3.23.4  / 3.23.8 )

And I keeps getting error message when I check "dmesg":

    ath10k_ahb a000000.wifi: failed to fetch board-2.bin or board.bin from ath10k/QCA4019/hw1.0

fortunately, someone on the forum give me the right direction
link: https://www.arednmesh.org/content/setup-hap-ac3-32340-problems

and I can verify it's the same issue I'm having.
link: https://forum.openwrt.org/t/no-wireless-mikrotik-rbd53ig-5hacd2hnd/157763

and here is that pull request from OpenWrt (still pending) :
https://github.com/john-tho/openwrt/pull/5
(I guess it have something to do with OpenWrt doesn't know how to unpack LZ77 data.)

so I pull AREDN from github, find the openwrt dir, add/modify the decompress method from @john-tho  and set config.mk to the right target.
(got a little hiccup when building, but it's not relevant)

finally, I have a working AREDN firmware for hAP ac3.

--
I want to thanks people on the forum / facebook group / github 
your advice really helps a lot.

Special shout-out to @john-tho , this won't be happening without his PR

so for people having latest version of MikroTik hAP AC3, there is a way to get your device working.

### And to save you some time, here is the firmware I built:

I really like @john-tho 's comment so I quote:
> usual disclaimer: expect your device to halt and catch fire…

[aredn-main-3635e22-ipq40xx-mikrotik-mikrotik_hap-ac3.zip](https://github.com/aredn/aredn/files/12474960/aredn-main-3635e22-ipq40xx-mikrotik-mikrotik_hap-ac3.zip)

SCWhite, 2023-08-30:
HI, @SensorsIot 
https://github.com/aredn/aredn/issues/813#issuecomment-1545493406
not sure if you're having the same issue with the new ac3 ?

SensorsIot, 2023-08-30:
I did not try. But there are rumors that some AC3s have a different radio that is not yet supported by OpenWRT. So I decided to stop my experiment. Plus: I switched my tunnel server to a virtual machine on Proxmox. So I do not need the second ac3

aanon4, 2023-08-30:
Thanks for this. This may solve problems we started to see on other, newer, Mikrotik devices. I should be picking up a known problem node tomorrow and will try this out.

aanon4, 2023-08-31:
So this fix will be in the next nightly build (20230831). If you have a moment to test it that would be fantastic.

SCWhite, 2023-08-31:
Definitely will give it a test, thanks !

aanon4, 2023-08-31:
I tested this and while it seemed to work just fine after the upgrade, after I power cycled this morning, the device no longer visible on my DtD connection. Have to investigate further today. Anyway .. might not want to test this. I take it you had no issues with your version? Did you only try it on known problematic hap ac3s?

SCWhite, 2023-08-31:
Well, this is my first AREDN device, so I'm not sure if it's normal for "Neighbor Status" takes like 5~10 minutes from "pending" -> "active"  but I do run through multiple power cycle, seem fine to me.

and I do recall there is one time "wifi scan" just can't find any device, but once I reboot both device, everything connect again.

also, I'm running v3.23.8 on my old ac3 + two "self built FW" new ac3
haven't flashed the old one with my version, should have give it a try.

BTW, when you say "DtD connection" is it connect directly with Ethernet cable or you mean through mesh RF ?

aanon4, 2023-09-01:
The pending -> active thing is entirely normal and how that bit works (it takes time to gather data about the state of connections).

aanon4, 2023-09-01:
So after the nightly build hosing my ac3 today .. I cant reproduce it at all. Reflashing my AC3 to nightly and upgrading went fine and nothing I can do makes it fail. This is less of a surprise than this change causing it to fail if I'm honest - that seemed so unlikely. But I wasn't the only one with issue this morning, so it just seemed a bit odd.

For the moment I've pulled the patch from the build so tomorrows nightly will be without it. I expect to return it once I've testing it a whole bunch more.

SCWhite, 2023-09-01:
HI, @aanon4 
I've try a few different combination, (i.e. all three with my build / all with nightly build 20230831 / mix of both ) with multiple power cycle. 
All of them are able to see each other, and I didn't think I run in any weird behavior. 

I'll keep on trying them with different setting, and see if any thing will cause them to break.

Meanwhile, is there anything I can test for you or anything that can help this issue?

closed 2023-09-05 by aanon4