aredn/aredn #921: 3.2.8.0 has same problem with using 5ghz as mesh on Mikrotik hAP

adzimme, 2023-08-31:
I posted an issue against the nightly build that I was using to allow me to select 5GHz for mesh and 2GHz for WAN.  When I upgraded to the new "stable" version, the same problem is present.  If you select 5G for MESH, it corrupts both /etc/config/wireless and /etc/config/network such that the wan interface is disabled.  

In /etc/config/wireless, the UI writes the file as follows :
config  wifi-device 'radio1'
      option type 'mac80211'
      option disabled '1'
      option hwmode '11g'
      option htmode 'HT20'
      option path 'platform/ahb/18100000.wmac'

config        wifi-iface
      option ifname 'wlan1'
      option device 'radio1'

config        wifi-device 'radio0'
      option type 'mac80211'
      option disabled '0'
      option channel '139'
      option chanbw '10'
      option country 'HX'
      option distance '0'
      option hwmode '11a'
      option htmode 'HT20'
      option path 'pci0000:00/0000:00:00.0'

config        wifi-iface
      option ifname 'wlan0'
      option device 'radio0'
      option network 'wifi'
      option mode 'adhoc'
      option ssid 'AREDN-10-v3'
      option encryption 'none'

config wifi-iface
      option ifname 'wlan0-1'
      option device 'radio0'
      option network 'wifi_mon'
      option mode 'monitor'

To fix the problem, change the /etc/config/wireless as per below:
config        wifi-iface
      option ifname 'wlan1'
      option device 'radio1'
      option network 'wan'
      option mode 'sta'
      option ssid '<ssid>'
      option key '<key>'
      option encryption 'psk2'

additionally /etc/config/network has to be modified.  The UI writes the WAN as being wlan0.  This needs to be changed to wlan1.

Hopefully since this made it into the stable build, it can be fixed.  Any network changes will cause the errors to be re-written.

SCWhite, 2023-09-05:
HI,
Is it a hAP2 or hAP3 ?
I just flashed some hAP3 and I've got no problem 5G for MESH.
if you have more detail, I probably can try it on my device. 

just a hunch, maybe do a clean reset will help? or try latest nightly build ?

adzimme, 2023-09-05:
Hap ac lite. Same problem on daily and stable build. I even regressed to the previous stable and came forward with the same problem. I posed all of the data in the previous version. I can again tomorrow. 

Sent from the all new AOL app for iOS

aanon4, 2023-09-05:
Please check nightly 20230905 to see if this fixes the problem.
NOTE: You must re-save your configuration from the Basic Setup page for the fix to take effect.

aanon4, 2023-09-16:
Fixed? @adzimme @SCWhite

adzimme, 2023-09-16:
Yes it does seem to be fixed. You have to save changes and reboot again but I’m now able to use 2.5g for wan and 5g for mesh without having to rebuild the config files. AlanSent from my iPhone

closed 2023-09-16 by aanon4