aredn/aredn #928: Advertised Services - Ampersand

n9jim, 2023-09-11:
I have a camera that needs a compound url using "&" (ampersand) to connect commands.
When placed in the advertised services, no link is created on the mesh status screen.

Looks a lot like this:
https://www.arednmesh.org/content/how-advertise-ip-camera-ip-stream#comments

Is this something that can be enabled?
Thanks,
Jim
N9JIM

aanon4, 2023-09-12:
While we were allowing these to be set in the config file, OLSRD was refusing the propagate them. This is fixed with this change.

Note that the service will only be seen by nodes running this update (old nodes still consider these paths illegal).

aanon4, 2023-09-16:
Fixed? @n9jim

n9jim, 2023-09-16:
TimLoaded nightly on the N9JIM-hAP2 (lite).Advertises correctlyLink worksLook at mesh status on n9jim-blackmtn-r5ac-90sect-68 (running 2.23.8.0) no advertisement N9JIM-r5ac-1 loaded with nightly with computer on it, no dtd to hAP, and RF connected thru “blackmtn” shows advertisement and link works.Summary….It works, And advertisement information passes thru nodes that are not nightly (but not displayed).ThanksJimSent from my iPhone

aanon4, 2023-09-16:
Yes.

closed 2023-09-16 by aanon4