aredn/aredn #943: Add field for azimuth

CoolAEW, 2023-09-20:
Hello!

Under optional settings, where you enter the latitude and longitude of the antenna, could you add a field for antenna azimuth?

This can help us identify antennas that are close together and map servers could use the data to visualize it.
For example:
[image]

ab7pa, 2023-09-26:
Not sure what this would be used for.  Adding a field on the protected UI means that no one but the node owner could view it.  Also, to me the azimuth is not very valuable unless you also know the beamwidth of the device.  Maybe you could elaborate on what you are trying to accomplish with the azimuth field?

[Many mapping programs](https://github.com/kn6plv/MeshMap) already calculate the azimuth automatically (example attached).

[image]

aanon4, 2023-09-26:
I'll admit, I've always wanted to azimuth field too, but never been quite sure how to do it well. I've wanted it so when people are setting up new nodes, they have some understanding about whether an antenna is pointing towards them, which seems useful. But I agree, without understanding the beamwidth too, it's not as useful.

Now for many radios with built-in antennas, we actually know the beamwidths, but for some we don't and you'd have to enter it. Not a big issue, but makes the implementation a little more involved.

ab7pa, 2023-09-26:
If the request is for fields to manually enter azimuth / beamwidth, it seems like an additional step would be required -- make those new fields queryable via the sysinfo.json api?

CoolAEW, 2023-09-26:
I was thinking just a simple manual field for Azimuth, that would be passed to the sysinfo.json api.
I see no problem with displaying the azimut in the same way as location on the status page aswell.

I've been working on making KN6PLV's meshmap display azimuth based on data from sysinfo.json and this is how I could see it being used, just a simple arrow/triangle to show where the antenna is pointing. Here I just add a transparent image with one corner opaque behind the node icon and rotate it based on azimuth.
[image]

For me this would be a nice feature to see what nodes are pointing in my direction when adding new nodes

ab7pa, 2023-09-26:
Since Tim KN6PLV already calculates the azimuth in his mapping software, he could probably add a small triangle without too much effort, if that's what you're asking for.

[image]

CoolAEW, 2023-09-26:
That only calculates the direction that particular link is coming from and not where the antenna is actually pointing. When there are multiple links you get multiple angles displayed, some of those links could even come from a side or back lobe

aanon4, 2023-12-18:
Done

closed 2023-12-18 by aanon4