aredn/aredn #956: Consider a shallow clone of OpenWRT sources for builds

hickey, 2023-10-10:
I am currently sitting in the Atlanta airport and trying to build a firmware image to test against. Early on in the build process a full clone of the OpenWRT sources is performed which takes quite some time on sluggish network connections. 

Since 99% of the time only the latest revision of the sources is needed or wanted, I would like to suggest that the clone command in the `Makefile` be changed to include `--depth=1` so that a shallow clone is performed. It is actually a little bit more than that. Here is the actual command to pull out a tag or branch:

```
git clone --depth=1 https://github.com/openwrt/openwrt.git --branch v22.03.5 --single-branch
```

That command works (significantly faster) for cloning the OpenWRT repo. On the current connection I was able to clone the repo in about 8 seconds vs. 54 seconds for the full clone. A few mins ago while building in the docker container the full clone seemed to take about 4 mins. Even at 54 secs a 7 times decrease in the cloning is significant. 

The only time that this change would not work is if `OPENWRT_COMMIT` is set to SHA1 hash for the check out. I am not sure if that is a use case that needs to be supported.

aanon4, 2023-10-11:
Could you let me know if this works for you?

https://github.com/aredn/aredn/pull/958

hickey, 2023-10-13:
Sorry for the delayed response. I am now in the motorhome making my way back across the country. 

This is effectively what I did in the local build directory to get the builds running faster. I did not care about the detached head warning, but that is a nice touch. I also left the `cd` in there as I figured that `.stamp-openwrt-cleaned` should be in the `openwrt` directory and that `.stamp-unpatched` would be found there. 

```
$(OPENWRT_DIR): .stamp-openwrt-removed
        git clone --depth=1 $(OPENWRT_SRC) $(OPENWRT_DIR) --branch=$(OPENWRT_COMMIT) --single-branch
        cd $(OPENWRT_DIR)
        ln -sf $(TOP_DIR)/patches $(OPENWRT_DIR)/
        ln -sf $(TOP_DIR)/files   $(OPENWRT_DIR)/
        touch .stamp-openwrt-cleaned
        rm -f .stamp-unpatched
```

aanon4, 2023-10-13:
Thanks. However I think the "cd .... " essentially does nothing as each line of a makefile is executed in a new shell, so that wont carry over to the next lines. In the old "git clone" it was used on the same line rather than specifying a target directory for the clone.

closed 2023-10-25 by aanon4