aredn/aredn #964: Support for CPE510 version 3.20?

kosmas12, 2023-10-17:
Good evening!

I want to get AREDN software running on some TP-Link CPEs I got, but the website lists support for revisions up to 3.0. Both my CPE510s are version 3.20. Will the v3.0 firmware work on v3.20? If not, how can I contribute and modify the codebase to get it working?

Thank you in advance!

aanon4, 2023-10-17:
It might just work, but without trying it I'm not sure there's anyway to know. Make sure you can reflash the original firmware in case it doesn't.

kosmas12, 2023-10-17:
Hello and thank you for the quick response!

In case it doesn't work, would you be able to please assist me with the steps I need to take to port to the new hardware revision? Thank you.

aanon4, 2023-10-17:
Unfortunately without a device on hand that's difficult. Are you familiar with porting OpenWRT to new devices already; or something similar?

kosmas12, 2023-11-03:
Hi and have a nice November! Apologies for this huge delay, it's been a bumpy ride...

I am not familiar with porting OpenWRT or anything similar (or, really, any type of firmware) on new devices. I also have not tried installing into my CPE510 yet because I don't have them on hand right now.

Again, sorry for the delay and thank you a lot for replying!

kc2cbd, 2023-11-05:
The V3 is supported on OpenWRT 23.05 and by the page on OpenWRT should have the same specs as the other versions.

aanon4, 2023-11-05:
I expect to roll 23.05 into the nightly at the beginning of December.