aredn/aredn #966: Nightly Build fails Winlink Express version check

sirrated, 2023-10-22:
I just upgraded my hAP from 3.23.8.0 to 20131018-d6f6c4b (nightly).  Under Telnet Post Office session in Winlink Express, when attempting to update the AREDN MESH Node List, I receive the following popup error:

Error getting master node info

Unable to get info from master node
Your computer must be connected to an AREDN MESH node
running firmware version 3.17.1.0RC1 or later.

Let me know if more info is needed!

Orv, 2023-10-23:
That's likely a Winlink Express error; looks like it's not recognizing the nightly build format.  I'll check mine and confirm it.

aanon4, 2023-10-23:
Could you provide some screenshots of the error, plus information about your browser and host computer?  Thanks.

Orv, 2023-10-23:
I'm running 20231013-a8c6ac9.
Here's the response that I get and it's normal.
[image]

sirrated, 2023-10-23:
Screenshot:
[image]

Local host computer: Windows 10 Pro, Firefox 118.0.2, connected to hAP via 5 gHz wifi.

sirrated, 2023-10-23:
I solved my issue.  My hAP had run out of DHCP addresses (5 host Direct).  Once my Windows PC had a an IP address, I did not have any issues updating the list of Telnet Post Office servers.  Thank you!

closed 2023-10-23 by sirrated