aredn/aredn #969: Nightly Build 20231018 not compatible with Rocket M9

wa3lab, 2023-10-28:


aanon4, 2023-10-28:
Please provide more details - thanks.

wa3lab, 2023-10-28:
I attempted to upgrade to the 20231018 nightly build via the sysupgrade file.  All appeared to go OK with LEDs flashing as usual and a reboot, but I was not able to connect to the Ethernet connection again.  I was also not able connect via RF nor does it show up on the mesh status on a different node.  I tried to flash the 20231018 factory firmware(via TFTP)  and it goes through the process but never completes.  I tried earlier factory versions and it will not even respond to the TFTP and gets a timeout waiting for the server to respond.

aanon4, 2023-10-28:
Were you running the last release before doing this upgrade? If not, what version, and can you restore to that?

wa3lab, 2023-10-28:
No.  It was running a version 3.19.3.0.  I tried to do a factory install of that firmware but the TFTP client just times out.  Maybe I'm not using the correct firmware.

aanon4, 2023-10-28:
Well we never touch the tftp bootloader, so not sure what's happening there. I dont have any 900 MHz hardware so this is impossible to test.

wa3lab, 2023-10-28:
Can you tell me where to get the proper factory file for 3.19.3.0 for the Rocket M9?

aanon4, 2023-10-28:
You can find the old builds here - http://downloads.arednmesh.org/firmware/ubnt/ - but I dont know which specific firmware the Rocket M9 requires.

wa3lab, 2023-10-28:
There was a table on the AREDN website that told you what file to use for your particular hardware.  Maybe I can use Wayback machine and get it that way.

VA2XJM, 2023-10-28:
hardware type: ubnt (rocket-m)  ATH79

;)

Jean-Michel Vien - VA2XJM
Président, Association des Radioamateurs de l'Abitibi-Témiscamingue
https://www.ve2atu.org
***@***.***
Ligne Association: (873) 476-0132 #822956

wa3lab, 2023-10-30:
UPDATE: By going back to aredn-3.18.9.0-ubnt-rocket-m-factory.bin I am able to flash and acquire the NoCall screen.  After setup the node connects fine on RF and DTD..  I can do a system upgrade to 3.19.3.0 and everything is good.  When I do a system upgrade to 3.20.3.0 the node flashes but it doesn't connect after it is done.  I used Wayback Machine .org to see what file the AREDN web site had listed for the Rocket-M M9 and it was aredn-3.20.3.1-ath79-ubnt_rocket-m-sysupgrade.bin.  I believe this is where the transition to ath79 started and apparently   broke the firmware for the Rocket M9. I found there are AR71xx files all the way up to 3.22.12 for the Rocket-m M9 which I have loaded and appears to connect via RF and DTD. So I'm guessing technically  , this is not an issue with 20231018 firmware.