aredn/aredn #983: GL iNet Mudi GL-E750 not taking AREDN firmware?

km6lslf, 2023-11-26:
I have an application for which the GL iNet GL-E750 looks like a good fit, but it doesn't seem to accept the AREDN firmware when I tried to upgrade it. Is there a trick to this device? Or is it not in fact compatible?

aanon4, 2023-11-26:
The support information indicates this device is untested - so a build exists but no one has ever run it (or reported back on running it). Can you provide more feedback?

km6lslf, 2023-11-26:
If you can provide instructions and let me know what outputs you're looking for, I'll provide any information possible. I can also send you this unit if you'd like to look at it directly.

aanon4, 2023-11-26:
I dont have a device, so instructions wont help me I'm afraid.  You said it didnt seem to accept the firmware when you tried. What did it do exactly? Screen shots of the failure might be helpful.

km6lslf, 2023-11-26:
Using the regular GL iNet process, I initiated the upgrade to the latest AREDN nightly build. It went to the installation screen in the attached image, which counted up to 100% without issue. 
[image]

From there, the expected behavior is for it to broadcast an open WiFi network for further configuration.

km6lslf, 2023-11-26:
Instead, it doesn't broadcast any WiFi and the display says System Updating. It doesn't change in a time frame I would expect. It's in that status now.
[image]

km6lslf, 2023-11-26:
No change after 40 minutes. I'll power it down. Tim, are you able to troubleshoot it if I send it to you? If so, let me know if you'd like me to reinstall the GL iNet factory firmware first.

aanon4, 2023-11-26:
Does it respond to the 192.168.1.1 address on any port?

km6lslf, 2023-11-26:
When in uboot mode, yes, but the default firmware IP is 192.168.8.1, and it
never issued a valid IP address when the display said System Updating

km6lslf, 2023-11-27:
I put the latest GL iNet firmware on the device and can send it to you if you have time to look into it further.

km6lslf, 2023-12-12:
I tried the latest stable release and got the same result. I then tried 3.23.04, which I believe is the first stable release that supported the Mudi and got the same result again. Has anyone ever loaded AREDN onto one of these? If not, it was probably premature to add it to the supported device matrix.

aanon4, 2024-01-23:
https://github.com/aredn/aredn/pull/1079

closed 2024-03-05 by aanon4

dracode, 2024-04-05:
I tried `3.23.12.0` on an E750. I experienced the exact same symptoms. It does not appear to work, sadly. Suggest this issue be re-opened.

aanon4, 2024-04-05:
The bug fix noted here is post 3.23.12.0. Please try nightly.