aredn/aredn #986: DNS lookups refused for local 802.11 clients

slightlyunconventional, 2023-12-01:
firmware version: 3.23.8.0

My network topology is:
```
mesh
└ ki7kkm-mobile (LiteBeam 5 AC)
  ├ element (laptop)
  └ ki7kkm-services (server)
```
All links are wireless; the LiteBeam is running an 802.11 access point and the server and laptop are connected to its SSID.

My problem seems to be that dnsmasq is refusing all DNS lookups from the laptop for the server. dig succeeds for the AREDN node, but not the server:
```
[hollisb@element ~]$ dig ki7kkm-mobile.local.mesh

; <<>> DiG 9.18.20 <<>> ki7kkm-mobile.local.mesh
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: NOERROR, id: 19307
;; flags: qr rd ra; QUERY: 1, ANSWER: 1, AUTHORITY: 0, ADDITIONAL: 1

;; OPT PSEUDOSECTION:
; EDNS: version: 0, flags:; udp: 65494
;; QUESTION SECTION:
;ki7kkm-mobile.local.mesh.	IN	A

;; ANSWER SECTION:
ki7kkm-mobile.local.mesh. 0	IN	A	10.30.201.236

;; Query time: 41 msec
;; SERVER: 127.0.0.53#53(127.0.0.53) (UDP)
;; WHEN: Fri Dec 01 15:19:32 PST 2023
;; MSG SIZE  rcvd: 69

[hollisb@element ~]$ dig ki7kkm-services.local.mesh

; <<>> DiG 9.18.20 <<>> ki7kkm-services.local.mesh
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: REFUSED, id: 47623
;; flags: qr rd ra; QUERY: 1, ANSWER: 0, AUTHORITY: 0, ADDITIONAL: 1

;; OPT PSEUDOSECTION:
; EDNS: version: 0, flags:; udp: 65494
;; QUESTION SECTION:
;ki7kkm-services.local.mesh.	IN	A

;; Query time: 23 msec
;; SERVER: 127.0.0.53#53(127.0.0.53) (UDP)
;; WHEN: Fri Dec 01 15:19:45 PST 2023
;; MSG SIZE  rcvd: 55
```

tcpdump on the laptop very simply shows:
```
14:06:13.101103 IP element.49057 > localnode.local.mesh.domain: 59608+ [1au] A? ki7kkm-services.local.mesh. (55)
14:06:13.105863 IP localnode.local.mesh.domain > element.49057: 59608 Refused 0/0/1 (61)
```

Adding an alias in the web UI for the server actually works.

I'll attach a few files that seem relevant.

slightlyunconventional, 2023-12-01:
[image]

slightlyunconventional, 2023-12-01:
[hosts.txt](https://github.com/aredn/aredn/files/13532666/hosts.txt)
[hosts_olsr.txt](https://github.com/aredn/aredn/files/13532667/hosts_olsr.txt)
[dnsmasq.conf.txt](https://github.com/aredn/aredn/files/13532668/dnsmasq.conf.txt)

aanon4, 2023-12-02:
Could you add the support tool data dump (there's a link at the bottom of the admin page)? Also, a screenshot of the ports page would be super helpful.

slightlyunconventional, 2023-12-02:
[supportdata-KI7KKM-MOBILE-202312020032.gz](https://github.com/aredn/aredn/files/13532826/supportdata-KI7KKM-MOBILE-202312020032.gz)

slightlyunconventional, 2023-12-02:
[image]

slightlyunconventional, 2023-12-13:
For what it's worth, I still see this behavior with the 20231212-ee0dd54 snapshot.