aredn/aredn #998: RF links that should be usable stuck in idle status?

km6lslf, 2023-12-07:
[KA6TCR-TCR-DPNIC-23 WiFi scan.pdf](https://github.com/aredn/aredn/files/13596955/KA6TCR-TCR-DPNIC-23.WiFi.scan.pdf)
### Describe the bug
RF links on a Rocket 5AC Lite are stuck in idle status and never seem to be added to a routing table. 

- [ ] Does this issue still occur on the latest version? (**Required**)
Started on recent nightly before 3.23.12.0 and persists with the latest stable release installed.
### Expected behavior
Successful RF connections when SNR is above thresholds set in LQM and signals are seen on WiFi Scan.

### Screenshots
Please include a screenshot of your system information (if possible) if the specific system environment is relevant to the bug.
WiFi Scan, Mesh Status and Neighbor Status pages are
[KA6TCR-TCR-DPNIC-23 Neighbor Status.pdf](https://github.com/aredn/aredn/files/13597026/KA6TCR-TCR-DPNIC-23.Neighbor.Status.pdf)
 attached in PDF form. 
[KA6TCR-TCR-DPNIC-23 mesh status.pdf](https://github.com/aredn/aredn/files/13597053/KA6TCR-TCR-DPNIC-23.mesh.status.pdf)

### Additional context
Add any other context about the problem here.

km6lslf, 2023-12-07:
They don't show up on the mesh page, which I believe is a symptom of the problem of their status being idle. In my experience, an SNR of 19 is definitely usable and an SNR of 14 is generally useful for low-bandwidth activities. The node ending in 226 is Krum, which reported a 100% LQ from KA6TCR-TCR-DPNIC-23.  The node ending in 173 is on the roof of my house and reports a 89% LQ. These suggest to me that these links should be usable.

aanon4, 2023-12-07:
The mesh page would show the link regardless of idle status - the fact that it's not routing traffic there doesnt effect OLSR which provides that data - rather the opposite really. What do the other radios think regarding the connection to this node?

km6lslf, 2023-12-07:
They show the LQ values described above and NLQ of 0. If you're looking for other information please let me know how to find the information you're looking for.

aanon4, 2023-12-07:
And the neighbor page information?

km6lslf, 2023-12-07:
Based on your information saying it was sending info to the DtD connection, I added a tunnel connection to my house to the hAP that  KA6TCR-TCR-DPNIC-23 is connected to. I now get the following information:

 KA6TCR-TCR-DPNIC-23 WiFi Scan:
<html>
<body>

SNR | Signal | Chan | Enc | SSID | Hostname | MAC/BSSID | 802.11 Mode
-- | -- | -- | -- | -- | -- | -- | --
27 | -82 | 131 |   | AREDN-10-v3 | - | 44:D9:E7:6A:8B:FB | Connected Ad-Hoc Station
27 | -82 | 131 |   | AREDN-10-v3 | KA6TCR-RM5120-KRUM-226 | 24:A4:3C:FC:CA:E2 | Connected Ad-Hoc Station
21 | -88 | 131 |   | AREDN-10-v3 | KM6SLF-NSM5-NORTH-173 | F4:92:BF:48:B6:AD | Connected Ad-Hoc Station

</body>
</html>

Neighbor Status:
NeighborLinkSNRDistanceQualityStatus  
[ka6tcr-hap-dpnic-208](http://ka6tcr-hap-dpnic-208.local.mesh:8080/)DtD-0.0 miles100%active
[ka6tcr-rm5120-krum-226](http://ka6tcr-rm5120-krum-226.local.mesh:8080/)RF19/124.1 miles92%idle
[km6slf-nsm5-north-173](http://km6slf-nsm5-north-173.local.mesh:8080/)RF13/93.3 miles60%idle
44:D9:E7:6A:8B:FBRF20-100%idle

It's now seeing Krum on the Mesh Status screen, but with 0 LQ and 94 NLQ. Still not seeing km6slf-nsm5-north-173

km6lslf, 2023-12-07:
Would it not be seeing km6slf-nsm5-north-173 because it's on the remote node list due to the tunnel connection?

km6lslf, 2023-12-07:
Krum mesh status showing DPNIC-23
[image]

And DPNIC-23 mesh status showing Krum.
[image]

km6lslf, 2023-12-07:
Here's the Neighbor status of the DPNIC-23 node:
[image]