aredn/aredn #999: Schedule of firmware file updates

CoolAEW, 2023-12-07:
### What is your question
Describe your question/problem as clearly as possible.

How often do you update the files available for download on the website?
I'd like to try out some of the new features that was merged earlier this week, but they don't seem to be live on the latest firmware batch. The files available are from 04.12.23

### Links and screenshots
Please include any relevant links or screenshots, if applicable.

http://downloads.arednmesh.org/firmware/

### Additional context
Add any other context about the problem here.

aanon4, 2023-12-07:
Are you referring to some PRs that were pending until yesterday?

dman776, 2023-12-07:
Files are available via downloads.arednmesh.org/afs/wwwThanks,Darryl

http://downloads.arednmesh.org/firmware/

CoolAEW, 2023-12-08:
Ahh, ok.
Then I misunderstood how the releases worked.
I'll just wait for the next nightly 😊

closed 2023-12-09 by CoolAEW