aredn/aredn #1006: Add DHCP options in GUI

CoolAEW, 2023-12-10:
**Is your feature request related to a problem? Please describe.**
We use Cisco IP phones that connect to a central FreePBX instance, for the phones to provision correctly we have a TFTP server with config files. To get this to work properly we have to SSH in to each node and add option 150 with the IP of the PBX. This config gets lost on fw updates.

**Describe the solution you'd like**
An way somewhere in the gui to add custom DHCP options.

**Describe alternatives you've considered**
Manually SSH into nodes and add DHCP options

**Additional context**
`echo "dhcp-option=150,xx.xx.xx.xx" >> /etc/dnsmasq.conf`
`/etc/init.d/dnsmasq restart`

aanon4, 2023-12-11:
Would an extra file you could add options to, and which is preserved across reboot and installs, suffice for now?

aanon4, 2023-12-11:
Such as something like this https://github.com/aredn/aredn/pull/1007

CoolAEW, 2023-12-11:
That would be great to have, but a way to add options through GUI later would also be nice 👍

aanon4, 2024-01-23:
https://github.com/aredn/aredn/pull/1073 might be what you're looking for.

closed 2024-03-05 by aanon4