aredn/aredn #1009: Corrupt mesh status current neighbor node name on TP-Link CPE 710 on 3.23.12.0

alchemy33, 2023-12-11:
Node name of TP-Link CPE 710 V1 appears incorrect on Mesh Status page of RF connected nodes.  The expected node name would be "KE3KQ-5G-ROFO" in the Current Neighbors field.  The displayed name is "dtdlink.KE3KQ-5G-ROFO /KE3KQ-5G-ROFO.local.mesh". This name is evident on nodes of Mikrotik SXT 2 & 5 GHz and GL-USB150 2 GHz

This was also like this in 10/18/23 nightly builds and also remains this way in current firmware version release 3.23.12.0

Expected behavior is the node name to be displayed on other nodes as "KE3KQ-5G-ROFO".  If dtd then this be appended after the node name and not as a prefix.  The "/" and repeat of the node local address seems related.  I suspect that this naming structure is related to other observations but this is the most obvious.  All this is RF, as I do not tunnel.  Installation of the current release firmware was uneventful.

I am attaching 2 images.  One is the basic node status screen of the TP-Link CPE710 v1.  The second is the mesh status page from an RF connect Mikrotik SXT node displaying the incorrect node name.

alchemy33, 2023-12-11:
Image of basic TP-Link CPE 710 v1 node screen.

[image]

Image of how node name is displayed on an RF connected node

[image]

aanon4, 2024-03-27:
https://github.com/aredn/aredn/pull/1128

aanon4, 2024-03-27:
This fixes the problem you're seeing. However, it's an install level fix so probably wont help you unless you reinstall the node from scratch.

alchemy33, 2024-03-28:
Wonderful!   I installed the CPE710 factory.bin and sysupgrade.bin nightly 3/27/24 files last night and can confirm the node name fix as noted in #1128.  Thanks Tim!

closed 2024-03-28 by alchemy33