aredn/aredn #1016: Services are not being advertised

kc2cbd, 2023-12-14:
### Describe the bug
A clear and concise description of what the bug is.
In the latest nightly Services are not being advertised.

- [ ] Does this issue still occur on the latest version? (**Required**)
 YES

### Expected behavior
A clear and concise description of what you expected to happen.
Services to be advertised

### Screenshots
Please include a screenshot of your system information (if possible) if the specific system environment is relevant to the bug.
[image]

aanon4, 2023-12-14:
Nightly?

kc2cbd, 2023-12-14:
Thanks.
Is there something I can do that will make it work for the time being?

aanon4, 2023-12-14:
Just go back to the release until tomorrow's nightly.

aanon4, 2023-12-14:
If you could verify that today's nightly fixes the services issue, that'd be fantastic.

kc2cbd, 2023-12-15:
Issue seems to be fixed

closed 2023-12-15 by kc2cbd