aredn/aredn #1020: New nightly builds don't show up when already on a nightly build

CoolAEW, 2023-12-14:
### Describe the bug
When trying to update from one nightly build to another no build shows up. I'll have to downgrade to a stable build before nightly builds shows up again.

- [ x] Does this issue still occur on the latest version? (**Required**)

### Expected behavior
When refreshing the available firmware list i expect the newest nightly build to show even when I'm already running a previous nightly build

### Screenshots
[image]

### Additional context
None

Orv, 2023-12-14:
I've found that it occasionally takes two or three tries for the latest nightly build to show up in the refreshed list.  But it eventually appears.

CoolAEW, 2023-12-14:
Ok, after 3 refreshes it showed up. But the refresh took a long time aswell.
[image]

aanon4, 2023-12-14:
I dont have a good explanation, at least when it comes to the code on the node and the code on the server. Neither has changed recently.

dman776, 2023-12-15:
I have been running the AFS updater manually while I figure out why the nightly build fails on that step.Thanks,Darryl

aanon4, 2023-12-15:
@dman776 It should be fixed and you shouldnt have to do that anymore.

closed 2023-12-18 by aanon4