aredn/aredn #1032: MikroTik RBLHG-5HPnD-XL antenna specs in error

Orv, 2023-12-19:
### Describe the bug
A clear and concise description of what the bug is.
Antenna gain and beamwidth for MikroTik RBLHG-5HPnD-XL are incorrect
- [ ] Does this issue still occur on the latest version? (**Required**)
Yes
### Expected behavior
A clear and concise description of what you expected to happen.

### Screenshots
Please include a screenshot of your system information (if possible) if the specific system environment is relevant to the bug.

If applicable, add screenshots to help explain your problem.

### Additional context
Add any other context about the problem here.
Gain should be 27 dBi; beamwidth should be 6.4 degrees

aanon4, 2023-12-19:
Do you want to post a screenshot? Because the data in the radio file reads 27/6.4

Orv, 2023-12-19:
[image]

Orv, 2023-12-19:
Very confusing discussion in the AREDN forums.   radio file is correct.

closed 2023-12-19 by Orv