aredn/aredn #1043: PowerBeam M5 400 firmware select issue, and upload issue

n9jim, 2023-12-22:
### Describe the bug
A clear and concise description of what the bug is.
On a Ubiquiti Powerbeam M5 400, Using installed firmware version 3.23.4.0: DOWNLOAD <refresh> firmware results in: no files found. Tried 2x.

Using a PC, I manually downloaded a file for Ubiquiti PowerBeam M5 XW (only possible one for 400?), and then used UPLOAD command, error: says firmware & hardware do not match. (tried to use the 3.23.12.0 2x, and then 3.23.8.0, all had same error.

I then retried the DOWNLOAD <refresh> again on the unit, and it found the files!?! and the unit updated. This occurred about 9:45PM PST.

Not sure why it eventually worked.  I don't currently have another unit to retry upgrading the firmware.
However, I tried it again on the same unit after the successful download, and it found the files with DOWNLOAD <refresh>.

- [ ] Does this issue still occur on the latest version? (**Required**)

### Expected behavior
A clear and concise description of what you expected to happen.
Find the hardware, load the firmware selection on the first try?

### Screenshots
Please include a screenshot of your system information (if possible) if the specific system environment is relevant to the bug.

If applicable, add screenshots to help explain your problem.

### Additional context
Add any other context about the problem here.

k1ky, 2024-01-07:
I have multiple units of this model running 3.23.12.0 and recent nightlies with NO firmware upgrade issues.

aanon4, 2024-03-29:
Any update on this?

n9jim, 2024-03-30:
I downloaded direct from the web to the device 3.23.12.0   (using the admin/download firmware)
then,
I manually went to the website and looked for the correct firmware to use.  The only logical choice seemed to be
Ubiquiti PowerBeam M5 XW
When I down loaded 3.23.12.0 for this device from the website to my computer, and then try to UPLOAD FIRMWARE into the node, the node responds with "firmware cannot be updated.  firmware file is not valid: firmware device match failed."

Jim

aanon4, 2024-03-30:
Looking at your specific node, it claims to be running rocket-m-xw firmware. This is probably a hold over from when quite a lot of Ubiquiti devices ran rocket firmware by default. 

The firmware you need is indeed the powerbeam m5 xw (I think think this is detailed in the supported devices file) but because you're running rocket firmware you'd have to force a dangerous update to load it.

n9jim, 2024-03-30:
interesting.. the device was still showing on the status page as powerbeam m5 400. there was no indication of the different firmware.  Thanks for looking!

Enabled dangerous updates & uploaded the powerbeam m5 xw 3.23.12.0 file from the computer, and it went fine.

Then uploaded the latest nightly "20240330..." and it uploaded and is running.
May need a cautionary statement for powerbeams. (I didn't see this info anywhere in the main aredn area or as a bug).

closed 2024-11-19 by aanon4