aredn/aredn #1044: Ubiquiti PowerBeam 5AC Gen2 failed to load

al0r-matt, 2023-12-22:
### Describe the bug
Attempted to load aredn-20231221-0b7db7f-ath79-generic-ubnt_powerbeam-5ac-gen2-squashfs-factory.bin today and the device has went unresponsive.

Started at WA.V8.7.11, downgraded to 8.7.0 and logged in fine. SCP'd the above firmware, executed the 5ac steps...

WA# hexdump -Cv /bin/ubntbox | sed 's/14 40 fe 27/00 00 00 00/g' | hexdump -R >
/tmp/fwupdate.real
WA# chmod +x /tmp/fwupdate.real
WA# /tmp/fwupdate.real -m /tmp/aredn-20231221-0b7db7f-ath79-generic-ubnt_powerbe
am-5ac-gen2-squashfs-factory.bin
Found  Active on[2] ...
Storing Active[2] ... [%100]
Writing 'kernel         ' to /dev/mtd2(kernel         ) ...  [%100]
Writing 'rootfs         ' to /dev/mtd3(rootfs         ) ...  [%100]
Done

Post reboot, nothing happens. Power cycling has no effect. No response on 192.168.1.1 or original IP.

- [ YES] Does this issue still occur on the latest version? (**Required**)

### Expected behavior
Device reboot and reachable via 192.168.1.1

### Screenshots
 ___  ___
|\  \|\  \      Welcome to airOS
\ \  \\\  \
 \ \  \\\  \    PRODUCT: PowerBeam 5AC ISO Gen2
  \ \  \\\  \   MAC: AC8BA9C8BA77
   \ \_______\
    \|_______|  VERSION: WA.v8.7.0

WA# hexdump -Cv /bin/ubntbox | sed 's/14 40 fe 27/00 00 00 00/g' | hexdump -R >
/tmp/fwupdate.real
WA# chmod +x /tmp/fwupdate.real
WA# /tmp/fwupdate.real -m /tmp/aredn-20231221-0b7db7f-ath79-generic-ubnt_powerbe
am-5ac-gen2-squashfs-factory.bin
Found  Active on[2] ...
Storing Active[2] ... [%100]
Writing 'kernel         ' to /dev/mtd2(kernel         ) ...  [%100]
Writing 'rootfs         ' to /dev/mtd3(rootfs         ) ...  [%100]
Done

### Additional context
Device started on WA.V8.7.11. Downgrade to WA.V8.7.0 was successful.

aanon4, 2023-12-22:
Thanks for this.
Just want to verify this worked fine using the current release build?

al0r-matt, 2023-12-22:
No, it did not work fine, thus my report.

Post reboot, nothing happens. Power cycling has no effect. No response on 192.168.1.1 or original IP.

Executing a hard reset also does nothing.

aanon4, 2023-12-22:
Thanks. Your report was using the nightly build. I was asking if you had tried the release build?

aanon4, 2023-12-22:
Welcome to the project. I'm sorry this is how it started. For future reference, nightly builds should be avoided unless you're happy working with testing code which is being actively developed and may - as you've unfortunately found out - not work. We try to make sure that's not the case, but there are a lot of hardware devices to test at any given moment.

Now .. to get you unstuck. What you have to do it reflash this device with the Ubiquiti firmware, and then start again to put AREDN on it.  To do this you need to put the device into recovery mode. I couldnt find a set of instructions for you specific hardware, but all Ubiquiti radios have the same process. Here are some general instructions - https://scoop.co.za/blog/ubiquiti-tftp-recovery - which I hope will help you.

al0r-matt, 2023-12-22:
Excellent. Recovered and loaded the December build without issue.

Thank you!

closed 2023-12-22 by al0r-matt