aredn/aredn #1050: 20240102-6d8528f: MikroTik SXTsq 5 ac (RBSXTsqG-5acD): No DHCP offer on ethernet port

nc8q, 2024-01-02:
Loaded Nightly 20240102 onto my SXTsq-5ac.
My workstation did not obtain an IP address via DHCP.
Loaded 3.23.12.0 and my workstation obtained and then it obtained an IP address via DHCP.

### Expected behavior
I expected my workstation to obtain an IP address via DHCP.
### Screenshots

[image]

If applicable, add screenshots to help explain your problem.

Workstation does not obtain reserved IP address.

aanon4, 2024-01-02:
Did this start happening on a particular nightly?

aanon4, 2024-01-08:
Closing due to no feedback to help debug.

closed 2024-01-22 by aanon4