aredn/aredn #1055: Reduced WiFi performance post-3.23.12.0 nightlies.

AJ6GZ, 2024-01-06:
### Describe the bug
Under nightly builds after 3.23.12.0 seeing poor RF link performance on PBM5 and SXT2 models.  Tested with several nightly versions up to 20240104.  Far side nodes running 3.23.12.0 (NSM2 XW) and 3.23.8.0 (RM5 XW, not my node).  Symptoms are much increased ping latency and dropped packets. LQM runs down to around 50% on the longer distance node (PBM5, 4.5mi) and has locked out a few times; normally ~85% LQM, 31dB SNR, and 100% LQ/NLQ.  SXT2 is a short distance (50ft) link on-site.  LAN side ethernet port pings and L2 switch stats are clean on both.

### Expected behavior
Same performance as release version.

### Screenshots
Nightly versions on the left.
[image]
[image]

### Additional context
[supportdata-AJ6GZ-1-PBM5-202401061838---3-23-12-0.gz](https://github.com/aredn/aredn/files/13850935/supportdata-AJ6GZ-1-PBM5-202401061838---3-23-12-0.gz)
[supportdata-AJ6GZ-1-PBM5-202401061908---nightly20240104.gz](https://github.com/aredn/aredn/files/13850936/supportdata-AJ6GZ-1-PBM5-202401061908---nightly20240104.gz)
[supportdata-AJ6GZ-1-SXT2-202401061838---3-23-12-0.gz](https://github.com/aredn/aredn/files/13850937/supportdata-AJ6GZ-1-SXT2-202401061838---3-23-12-0.gz)
[supportdata-AJ6GZ-1-SXT2-202401061908--nightly20240104.gz](https://github.com/aredn/aredn/files/13850938/supportdata-AJ6GZ-1-SXT2-202401061908--nightly20240104.gz)

k1ky, 2024-01-07:
Also very degraded performance with Rocket M5 - won't connect to a fairly good strength station several miles away.  Reverted back to 3.23.12.0 Production - returned to normal connectivity.

AJ6GZ, 2024-01-24:
Re-tested with build 202401024. Problem remains.

aanon4, 2024-01-24:
Any chance this is bench testable in some way? I obviously cant try to debug these nodes in the field.

Orv, 2024-02-18:
On -ac platform, throughput is noticeably worse on nightly build vs. production release code  (30 -> 22 Mbit/sec) .   
On 802.11n links performance with nightly is severely degraded on nightly compared to release. (32 -> 13 Mbit/sec)
Tested with native iperf3, command line.    Tested in 2/12/24 with nightly that was current then.

aanon4, 2024-03-25:
Could someone re-test the bench test which I think was using a SXT2 ? Assuming the problem persists, can you detail the exact setup you're using for this bench test (details of both ends, channel used, power settings, bandwidth)? I finally had chance to run the cabling to setup a distance test in a semi-permanent way, so I'd like to setup a system which shows the problem here.

aanon4, 2024-03-26:
https://github.com/aredn/aredn/pull/1125

Would be interested in feedback with tomorrow's build (20240326) which has increased the low-level retry limit. Some preliminary testing suggests this was causing the issue.

k1ky, 2024-03-27:
Improved MESH Wi-Fi performance confirmed with 20240326 on Rocket M5XW and PBEM5XW units.  Comparable to 3.12.12 results.

aanon4, 2024-03-27:
@k1ky Thanks for this. Tomorrow I've reverted a couple of changes so it would be good to get your input on how that feels vs. how today's build feels. Thanks.

k1ky, 2024-03-29:
N-20240328 Shows some improvement over 0326 for this issue. I believe we are back to comparable performance or better than last production 3.23.12.0

AJ6GZ, 2024-03-30:
Tested 20240327.  RM5 vs. PB400 and SXT2 vs NSM2. Both appear to have returned to normal performance!  Also tested 20240329 with same result..  
Will close this issue unless anyone has more comments...

aanon4, 2024-03-30:
All feedback appears positive, so closing.

closed 2024-03-30 by aanon4