aredn/aredn #1058: Unable to load meshchat v2.1-v2.9 package onto devices running AREDN v3.23.12.0; however, v2.0 will install.

convolved-KY4G, 2024-01-08:
### Describe the bug
Unable to load meshchat v2.1-v2.9 onto devices running AREDN v3.23.12.0; however, v2.0 will install.

Versions 2.1 through 2.9 give the error shown on the lines below.
Collected errors:
* pkg_init_from_file: Malformed package file
/tmp/web/upload/newpkg.ipk.

- [ ] Does this issue still occur on the latest version? Yes

### Expected behavior
I expected the package to load without errors.

### Screenshots
[image]

### Additional context
I tried this on two different GlL.iNet AR300M16 nodes and a Mikrotik LDF2 node with the same unsuccessful results.
meshchat_2.0_all.ipk loads fine.  Anything newer does not.

aanon4, 2024-01-08:
Should file this bug on MeshChat as it seems the ipkg is malformed.

closed 2024-01-08 by aanon4

convolved-KY4G, 2024-01-08:
I will file a issue on the MeshChat effort as you requested.  I posted it here as I thought the issue might be connected to the current AREDN release since all meshchat packages going back all the way to two years ago seem to be affected.