aredn/aredn #1063: In-Node Map Tile URL

r1de, 2024-01-14:
### Describe the bug
In the configuration when you click the "Show Map" button, you get no map tiles.

- [ X] Does this issue still occur on the latest version? (**Required**)

### Expected behavior
You see map tiles.

### Additional context
"Stamen Maps" is now "Stadia Maps" and costs $ beyond something like 200k tiles/month.
It is no longer free.

I would recommend just using good ole' OSM and changing the default "Map Tile URL" to:
`http://{s}.tile.openstreetmap.org/{z}/{x}/{y}.png`

aanon4, 2024-01-14:
Nightly?

r1de, 2024-01-14:
@aanon4 If you already fixed it then ignore me.
I have not been running nightly builds.
Another user pointed out to me that the tiles did not load and I offered to create the issue.

aanon4, 2024-01-14:
I think it's fixed in nightly. Be nice if someone other than me checked though :-)

r1de, 2024-01-14:
I asked one of the local nightly build junkies to check the URL, but it does look like you fixed it #982

aanon4, 2024-01-14:
Okay to close?

r1de, 2024-01-14:
Yeah, I think so.

closed 2024-01-14 by r1de