aredn/aredn #1075: Wireguard Client IP invalid

k6ccc, 2024-01-22:
I have four Wireguard tunnels set on on my tunnel server.  The first three are working properly, but the forth one is not.  That forth one is showing a client IP of 172.31.231.254:5528.  It appears that the WG tunnels are using a /31 address since each one is two IPs higher than the previous - so that may in fact be valid.  However as I said, it is not working.

It gets better however.  The next tunnel I set up is showing an Client IP of 172.31.231.256:5529.  Last time I checked, .256 is not valid at all in IPv4...

aanon4, 2024-01-23:
Can you confirm the problem is fixed in the nightly?

k6ccc, 2024-01-23:
Yes, Nightly 20240122 fixes the problem.

Jim

https://github.com/aredn/aredn/issues/1075#issuecomment-1905272177

closed 2024-01-23 by aanon4