aredn/aredn #1089: VMware/QEMU x86 - Advanced networking missing and no firmware updates due to mismatched board_id and board_name checks.

marx1, 2024-02-05:
### What is the hardware/platform
Vmware ESXI 7.0 update 3
Unraid 6.12.6

### Links
https://unraid.net
https://vmware.com

### Additional context

I have multiple servers, one runs Unraid (QEMU Virtualization) and I just setup a VMware 7.0 u3 on a Dell R530. 

When I had AREDN installed and running on the QEMU it was reporting an "UNSUPPORTED DEVICE" and wold not let me update as it could not find an image. I talked with Tim KN6PLV a little bit about it; but at the time it was just something I was playing with. On that server I had a directly-mapped network card so the VM ran in single interface mode. 

Fast forward to today - I'm in the middle of building a dedicated ESXi server (because of various reasons) on my Dell R530. I was able to migrate the image I had on the Unraid box to a VMDK, booted it, and wanted to add the additional interfaces to make it a NAT gateway for services running on the server.

I ran into issues with connecting, and ran down the rabbit hole of getting a temporary IP and could connect to the web interface - It no longer showed "unsupported device" but I couldn't configure the extra ports.  I dug into github issues/requests and found  #995 . 

After hours of searching/testing I found the potential cause of all of this - 

**For firmware updates** 
In /www/cgi-bin/admin line 290  https://github.com/aredn/aredn/blob/9e94c018c7a0807035fd60811cdbf6854bb0424b/files/www/cgi-bin/admin#L290

The board type is hard checked to be "qemu-standard-pc-i440fx-piix-1996" or "VMware, Inc. VMware Virtual Platform" AND the type must be "generic"

This is causes issues with QEMU - in . /etc/board.json and the scripts it uses to check in /usr/local/bin shows  Unraid shows "qemu-standard-pc-q35-ich9-2009"  This is due to  unraid the default is Q35 for linux, not i440.  

Unraid (qemu):
[image]

To fix this I think it should regex match "qemu" and "vmware". If you're running AREDN in vms the chances are you know what you're doing.

For the advanced networking, It's a similar issue:

in /usr/lib/lua/aredn/nav/admin/65advancednetwork.lua (https://github.com/aredn/aredn/blob/9e94c018c7a0807035fd60811cdbf6854bb0424b/files/usr/lib/lua/aredn/nav/admin/65advancednetwork.lua#L2)
we're hard coded to check for it i440fx agian.

same in /www/cgi-bin/advancednetwork line 64: https://github.com/aredn/aredn/blob/9e94c018c7a0807035fd60811cdbf6854bb0424b/files/www/cgi-bin/advancednetwork#L64

On the VMware side, for the advanced networking

For both of these in 65advancednetwork.lua line 1 we check call aredn.hardware.get_board_type(), this actually returns the "id line from /etc/boards.json: [function hardware.get_board_type()](https://github.com/aredn/aredn/blob/9e94c018c7a0807035fd60811cdbf6854bb0424b/files/usr/lib/lua/aredn/hardware.lua#L163)

But this is incorrect based on what text is hard set in /usr/lib/lua/aredn/nav/admin/65advancednetwork.lua (https://github.com/aredn/aredn/blob/9e94c018c7a0807035fd60811cdbf6854bb0424b/files/usr/lib/lua/aredn/nav/admin/65advancednetwork.lua#L2 - that's looking for "VMware, Inc. VMware Virtual Platform" not "vmware-inc-vmware-virtual-platform"

Vmware:
[image]

With all of this I see two problems -

1. The image will **always** default to the 1-port config, because it never matches stuff properly . I think this is the real cause of #995 
2. QEMU and VMware can change  the board ID - I know the dev team doesn't want to support every config for X86 out there. It's good to say that if you're running on x86 you're not the average AREDN user. 

Therefore I suggest - drop the board types for virtual/x86 platforms, and switch to a regex/match for "vmware" and "qemu"

This also plays into supernodes, running it on x86 hardware is probably going to be a good idea in the future;

marx1, 2024-02-05:
Looks pretty good, I wasn't sure exactly where it was all used; but this should help with things. Thanks!

aanon4, 2024-02-05:
Actually ... I'm going to make another pass at this, instead collapsing all the qemu-* and vmware* to simple qemu and vmware generics much earlier. Stay tuned.

Marx1, 2024-02-06:
I manually built the code today, I can confirm this is working as expected. Thanks!

[image]
[image]
[image]

One item I noticed is you can't set DtD to be untagged; so you still need to be able to pass vlans to the VM; most hypervisors can do this; but we may want to enable allowing it to be untagged on virutal machines. I'll put a FR in later for that. For now; it's working.

closed 2024-02-06 by Marx1