aredn/aredn #1100: Node configuration enhancement - auxiliary antenna inf0

Orv, 2024-02-23:
**Is your feature request related to a problem? Please describe.**
In the case of an AREDN node mounted at the feedpoint of a dish antenna  (E.g., LDF or Nanostation), there's no place in the nightly bu8ld to list the dish and resultant gain derived from the combination of radio & dish

**Describe the solution you'd like**
Would be nice to have a field to record those pieces of data

aanon4, 2024-03-05:
LDFs already have an external antenna - does it also have some gain we need to handle?
Nanostation - is this the only one which requires an extra external antenna? Are there others? I'd like to only add this for radios which make sense.

Orv, 2024-03-06:
Both LDFs and Nanostations have internal antennas with gain and directivity, but also have the option of being installed at the feedpoint of a dish.   Suggest generic "dish" field with fields for gain and directivity (suitable sanity checked).  Maybe a field for total gain also, the sum of the two.
I'm not aware of any others with these capabilities.

aanon4, 2024-03-06:
I'm reluctant to add free-form fields here. So far we've maintained a list of antennas that people can add to radios such as rockets. This was a deliberate choice so we can see what actual antennas people are using. I'd rather to the same here with a list of additional antennas people can add to radios which support them.

Orv, 2024-03-06:
Maybe just add canned  entries: this is typical of dishes for Nanostations: https://www.kpperformance.com/ubiquiti-m-series-radio-reflector-dish, which for 5 GHz typically adds 23-26 dBi.  Maybe a check box indicating that's how it's being used, and show the additional gain.
For the Light dish feed, it's intended for use at the feedpoint of a satellite dish, which you can assume is about the same gain as a reflector dish and could be treated that way.
https://www.cambiumnetworks.com/wp-content/uploads/2017/09/Spec_ReflectorDish_Nov2014206-3.pdf

aanon4, 2024-03-29:
I've added aux antenna support for all nanostations. What else? LDF is already supported.

n9jim, 2024-03-29:
Note the net gain for a radio/internal antenna feeding a dish is not the addition of the 2 gains. It's only the reflector gain.  For a dish this is directly proportional to the dish diameter:
https://www.everythingrf.com/rf-calculators/parabolic-reflector-antenna-gain
Jim

ae6xe, 2024-03-30:
The reflector specs include the gain of the device mounted.   In the cambium NS reflector specs the heading for gain is "(Integrated Antenna + Reflector Dish)".     In the math, the physical path of the energy, we have PA output + internal antenna gain + reflector gain.   energy in a copper wire, radiation out with 45 deg (H) beamwidth,  energy concentrated again to ~10 deg (H) beamwidth.   

Last time I looked (a few years ago), I could not find with the KP Performance reflector any gain in the specs from the manufacturer.   There are different mounts to buy for different devices, and the total gain would be different.    

Suggest  to capture in a table the selection of known device and antenna combinations, and hardcode the total gain #.  This comes from the spec sheet, or we have to determine (in the case of the NS PK reflector combination).

n9jim, 2024-03-30:
Joe, 
The Cambium PDF says:
TOTAL GAIN (Integrated Antenna + Reflector Dish)
Not to be confusing but it means Antenna & Reflector dish system.  It is not the sum of the 2 gains.
The proper gain for a dish is a formula based on the size of the dish. The feed gain does NOT add to the dish gain. Instead, the feed must properly illuminate the dish in order to get the dish's full gain potential, a number <100%.
Looking at the Cambium PMP & PTP 450 (5GHz) Total Gain spec, it says 25dBi.   This is about right for the full system (feed & dish).
The formula for the gain of a 450mm dish at 5ghz results in 26dBi (at 100% efficiency, 50-65% is typical for a well designed feed).
https://www.everythingrf.com/rf-calculators/parabolic-reflector-antenna-gain

I've used the NSM3 as a feed to satellite dishes in the past, the solution works, but not very well, as the NSM3 has 3 patch antennas in the unit. Only one of which is actually at the feedpoint for the dish, the other two are virtually useless. The effective gain is actually less than a dish with a single patch, because 2 of the feed patches take power but don't apply it to the dish. 

Maybe using the formula would be easier than a lot of different tables for combinations of feed & dish. There is also a formula for the beamwidth.  This link has both BW & gain:
https://www.everythingrf.com/rf-calculators/parabolic-reflector-antenna-calculator

Hope this helps.
Jim
N9JIM

ae6xe, 2024-03-30:
Let's look at this from the perspective of "what is the total radiated power" coming out of the reflector.  This is the practical data point needed to deploy.  How is that calculated?   (We best determine this here, and hard code, as not everyone will be able to do so.)

Let's start with the device set to 25dBm xmit power coming out from the wifi chip and Power Amp.   Consider the 5Ghz PMP 100 device.

Would we all agree the radiated power is calculated by?:

xmit power setting (25dBm) - cable loss (negligible inside this device) + internal antenna gain (7dBi per PMP 100 specs) - loss due to alignment + reflector gain (calculated at 26dBi) = radiated power

The Cambium specs for this reflector with the PMP 100 mounted states gain is 26dBi @ 5.8Ghz.   

What value do we define to have a good estimate of radiated power = (25dBm + x )?   

I'm interpreting such that the PMP 100 mounted to the reflector has a total gain of [internal antenna gain of 7dBi - loss due to alignment (must be -7dBi?) + reflector gain (26dBi)].    I propose, use x = 26dBi.

Joe AE6XE

n9jim, 2024-03-30:
X is specified in the Cambium document provided by Orv as Total Gain. (26dbi for the PMP100 & reflector dish)
https://www.cambiumnetworks.com/wp-content/uploads/2017/09/Spec_ReflectorDish_Nov2014206-3.pdf

The manufacturer calced/measured this for the system.

This matches the intent of the PTP100 datasheet which presents the dish as gaining an additional 19dBi. (19+7)
https://www.cambiumnetworks.com/wp-content/uploads/2017/09/Spec_PTP_100.pdf

This works for the set of Cambium products using this reflector dish.

A good reference for more details on dishes & feeds can be found here:
http://www.w1ghz.org/antbook/contents.htm

Jim
N9JIM

ae6xe, 2024-03-31:
"presents the dish [reflector] as gaining an additional 19dBi".   Which appears to conflict with the calculation you pointed out, "The formula for the gain of a 450mm [reflector] dish at 5ghz results in 26dBi ".    But it is explained from the formula above:   The 'x' gain we are looking for is  (antenna gain - loss between antenna and reflector + reflector gain) or (7dBi - 7dBi + 26dBi)     Cambium appears to be munging the loss and physical gain together to call it 19dBi (-7 + 26).

I think we agree that x=26dBi is the value to use from the above example.   To move towards an implementation,  we may not have the same spec from the PK Performance reflectors for NS or Microtek LGHs.  

If so, we're choosing between the lessor of the evils to move foreword:
A) estimate the gain for the given device and reflector to hardcode (in some cases we don't exactly know); or
B) implement a free-form "enter what you think the gain is"

@aanon4, your call :) . 

Joe AE6XE

closed 2024-08-18 by aanon4