aredn/aredn #1113: ED25519 SSH keys not recognized

hickey, 2024-03-07:
### Describe the bug

Noticed a few days ago that if a ED25519 SSH key is uploaded through the administrative settings the page indicates that the key was accepted and included in `authorized_keys`. Yet inspection of `authorized_keys` does not show the new key. 

To further confuse the issue, I did a bit more testing. I generated a new RSA test key and tried to upload that to the node. It also would not appear in the `authorized_keys` file. I deleted the existing RSA key and tried to upload the test key again and it would not be put in `authorized_keys`. I then uploaded the original RSA key and that would upload fine. 

I have attached below the ED25519 key that I was trying to upload, the test RSA key that I just generated and the original RSA key I was using that does work. That way one can debug with the keys that I was having issue with. 

- [x] Does this issue still occur on the latest version? (**Required**)

### Expected behavior

SSH key would upload to the node and be listed in `authorized_keys`. 

### Screenshots

[image]

### Additional context

ssh-ed25519 AAAAC3NzaC1lZDI1NTE5AAAAIO9/49Z2pQv0nxJfAY6cr3AI+FRqe7uK1Swr03Zhp+gZ loginkey-20231214

ssh-rsa AAAAB3NzaC1yc2EAAAADAQABAAABgQDgNKj2sRjJFyBuiPYip3bcmyleps+MPoEozH5l3SYV9p94cy4YOQKkluBtdX9fSpNqfp0w1TQCP6JSShoHuZIq+22X4z5CQNVBFePc2VzI24UgE+KfCFhT9Zn5Q0F3ErjIAL8jv/OFDgw0x0xeE8I2wYHZLTPzj2mWEofs4zqlKk8DRVRuGc8H5e/B4JZ+jGISWNbPhC0qVeMubYvblitDY+okh4/bz3g17GOqGtPChQftPhU86aPVbl4si+MPXavGSjfA2n0mEVopu05fZVXGW/vC/tmvqLjHYHvi7EoJ5tIeFf1StWzDK8pbdocxVlDlhig0cpOVJaCQ6D/jHj5Zh13BD0thnR03rF/TavnSDa4ELsJFxqOwqWCi1Pd+of1jBDMm3/X7MDMc1AxHINZbT2pwQGN6WX2mPbp8XyUm+6LUyBz/5gq+NsH4XNvoJivQyPlGxCEmKyG8qTvXvY1hmDc8eSTnzh9/7rx9EjhDZLCIHtKf6r0HpyfVICb5Uz0= testkey

Original key that works:

ssh-rsa AAAAB3NzaC1yc2EAAAABIwAAAIEAptqLu+3SyMkcRrXtrmIXBh4k2Ug8Wx8pTjMGNsZzC2CHhp6WJzpHgaln1mbgwdCRbTfwmak927jUZhH8ZEFQxEGFBS+IRMo8RRsrzAyq8hopH0Ob1RQNXykRXtUm5RwVQHAAvOzTUXxBmUktJI95GMOCXvZpzfvMYAFgSQlmbOs= hickey@jhadur

hickey, 2024-03-07:
I should have also clarified that I tested with the latest nigthly (20240306-2cda75e) and the problem was occurring with it.

aanon4, 2024-03-07:
Key are only valid if they have USER@SOMEWHERE at the end as your working key does, but the other keys do not.

hickey, 2024-03-07:
Why is that? Not trying to be defensive but at the end of the public key it is a comment field for identification of the key set not a specific format like `USER@SOMEWHERE`. In addition, the page should not come back and state that key was installed if in fact it was not. That in itself is a bug. 

In addition, there is nothing in the documentation that even references that the public key will only be accepted if the comment field is formatted as `USER@SOMEWHERE`. Most keys in the real world are not generated not with anything that looks like an email address. Why is there this artificial requirement on the key's comment field?

hickey, 2024-03-07:
Also that is if the key has a comment at all. There are key sets that are generated with no comment at all. Which is a perfectly valid key set.

aanon4, 2024-03-07:
> Why is that?

Absolutely no idea. It's always been that way. I'll have to ask whomever originally put that restriction in place.

closed 2024-03-30 by aanon4

sarcasticadmin, 2024-04-22:
I ran into this as well. At the very least the `USER@SOMEWHERE` restriction should be mentioned as an error on the page. Without this theres incorrect information reported back to the user.

Prior to uploading a key:
[image]

After uploading a key (rsa) that doesnt include a comment of `USER@SOMEWHERE`:
[image]

I still had to login via SSH using the root password and  `/etc/dropbear/authorized_keys` is empty:
[image]

---

As to why theres the restriction of `USER@SOMEWHERE`. It looks like it simplify the lookup of keys for removal: https://github.com/aredn/aredn/blob/d261ebe334a121d8fd90951c073652bf54aa3e97/files/www/cgi-bin/admin#L745-L763

There also isnt a mention of this requirement on https://docs.arednmesh.org/en/latest/arednGettingStarted/advanced_config.html#administration for the authorized_keys section. Ive opened an issue in the docs: https://github.com/aredn/documentation/pull/350