aredn/aredn #1116: CPE710 connects briefly then drops link

wa3lab, 2024-03-19:
### Describe the bug
CPE710 will boot up and briefly connect to a NanoStation M5.  After a few minutes the connection will drop and the CPE710 will display 100% LQ and 0% NLQ.  The M5 will show the CPE710 as a previous neighbor. 
I ran a WIFI scan the CPE 710 and it sees the M5 IP address.
I ran a WIFI scan on the M5 and it sees the CPE710 as dtd.link-WA3LAB-CPE710  
I ran charts from both nodes and both see the other nodes signal with >25DB SNR

- [ ] Does this issue still occur on the latest version? (**Required**)
- [ ] YES - Nightly build 20240316

### Expected behavior
To connect and communicate.

### Screenshots
[image]
[image]

### Additional context
The CPE710 had the same issue on 3.23.12.0.  This behavior has been seen by other hams as well. 
Support Data
[supportdata-WA3LAB-CPE710-202311141428.tar.gz](https://github.com/aredn/aredn/files/14655405/supportdata-WA3LAB-CPE710-202311141428.tar.gz)

aanon4, 2024-03-19:
Could you provide the support data for both end, taken as close to the same time as possible? Thanks.

aanon4, 2024-03-19:
Also, does this occur when using the current release builds (not nightly) on both ends?

wa3lab, 2024-03-19:
2 snapshots a few seconds apart.  Also I am pretty sure it happens when both are at release versions, although I was connecting the CPE710 to a PowerBeam M5.  I can try to get the CPE710 back to release version, but have had difficulty using PXE on this node. 

[supportdata-WA3LAB-CPE710-202311141457.tar.gz](https://github.com/aredn/aredn/files/14656085/supportdata-WA3LAB-CPE710-202311141457.tar.gz)
[supportdata-WA3LAB-NanoM5-202304272035.gz](https://github.com/aredn/aredn/files/14656088/supportdata-WA3LAB-NanoM5-202304272035.gz)

aanon4, 2024-03-19:
Could you make sure LQM is enabled on both nodes and repeat the experiment? Thanks.

wa3lab, 2024-03-19:
[supportdata-WA3LAB-CPE710-202311141533.tar.gz](https://github.com/aredn/aredn/files/14656521/supportdata-WA3LAB-CPE710-202311141533.tar.gz)
[supportdata-WA3LAB-NanoM5-202304272111.gz](https://github.com/aredn/aredn/files/14656523/supportdata-WA3LAB-NanoM5-202304272111.gz)

aanon4, 2024-03-19:
It looks like the CPE has only been up about a minutes when you took these logs. Can you run the test again, give it 10 minutes, and then dump the support data? Thanks.

wa3lab, 2024-03-19:
Here is a shot after 30 minutes.  Do you still want one at 10 minutes?

[supportdata-WA3LAB-CPE710-202311141601.tar.gz](https://github.com/aredn/aredn/files/14656745/supportdata-WA3LAB-CPE710-202311141601.tar.gz)
[supportdata-WA3LAB-NanoM5-202304272139.gz](https://github.com/aredn/aredn/files/14656746/supportdata-WA3LAB-NanoM5-202304272139.gz)

aanon4, 2024-03-19:
So the odd thing is that the radios seem to be connected, but it's not working otherwise. LQM appear happy. Would you willing to put the nightly build on your NanoM5?

wa3lab, 2024-03-19:
That's exactly what I saw.  Everything looks like it should work, but it's not.  I rebooted and was going to wait 10 minutes and take another shot, but this time it connected and stayed connected.  I can load the nightly on the Nano if you want.

aanon4, 2024-03-22:
I have a CPE710v1 in my test farm, so I'll try to replicate this on that. Wanted to verify you also have the v1 of this device?

wa3lab, 2024-03-22:
Yes I have a version 1.  My friend just dropped off his CPE710 and I am going compare the performance to mine.  I have rolled the firmware back to 3.23.4.0 and so far the "100% LQ and 0% NLQ" has not occurred on the CPE710.  I'll keep upgrading until I see it fail.  Thanks.

aanon4, 2024-03-27:
Getting back to this ...

So I booted up my CPE710 and put today's nightly on it. This has a new OpenWRT version, but as it has minor changes I'm not sure I'd expect much difference. 

But I can't replicate your situation. The CPE710 connects just fine (I'm using channel 182 @ 10MHz) and doesnt seem to disconnect from anything I do a wifi scan from. Any thoughts?

wa3lab, 2024-03-30:
Likewise getting back to this...  A few domestic emergencies have curtailed my radio activities.  I have setup my test on channel 182, 10 mhz bandwidth, and LQM off .  I have eliminated the original WA3LAB-CPE710 from the testing as I am slowly convincing myself there is a hardware issue with that unit.  I will continue with a loaner node WV3P-CPE710.  I did a FACTORY load of nightly build 20240327 (per conversation with Ben KE3KQ regarding issue #1009) and was pleasantly surprised to see the dtdlink problem had been resolved. Thank you! After Easter I'll get back to more testing , but initial results are as follows.  Using  WA3LAB-NanoM5(with connected laptop) as the connecting node, both the CPE710 (also with its own connected laptop) and the NanoM5 powered up and connected normally. (100% LQ/100 % NLQ).  I ran a WIFI scan from the NanoM5, and after a short pause the results screen came up with the normal expected response.  I hit the quit button and returned to mesh status , which appears the normal 100% LQ/100 % NLQ.  At this point if I click on the WV3P-CPE710 link from the Nano, it will time out in about 10 seconds indicating  WV3p-CPE710 took too long to respond.  At this point if I try to ping the CPE710 it will fail with no response.  From the laptop plugged into the CPE710, the mesh status indicates 100% LQ/100 % NLQ, and all the remote nodes show up, and everything looks ok.  If I click on the WA3LAB-NanoM5 link on the mesh status page, it will time out.  Also pings from the CPE710 will timeout.  Almost done...please bear with me.  If I go back to the NanoM5, I can successfully run a WIFI scan again, and it will complete normally, however the link and pings continue to time out.  So, everything looks normal, but the nodes cannot ping each other.  I have seen this same failure condition happen on powerup also.  This condition will persist until I reboot the CPE710. 
      Sometimes I can run the WIFI test OK once, but it will usually fail on the second time.  Please understand I am just using the WIFI test to aggravate this situation to possibly help in debugging.  As a final step, I powered on my WA3LAB-PB-5g node as a sanity check.   From the NanoM5, I can click the link and display the PB-5G web page and ping as well.  All pings and web page requests to the CPE710 fail from all other nodes. 
I realize this is somewhat different from the original failure, so if you want I can close this ticket and open a new one?  Sorry to be so verbose and thanks for what you do.
73, Glenn WA3LAB

aanon4, 2024-03-30:
This seems like a different thing so yes, let's close this and open a new bug for this new issue.

wa3lab, 2024-04-01:
Sounds like a plan.  Our local group owns 15 of the CPE710 model and are anxious to get them in the air. 
Glenn WA3LAB

closed 2024-04-01 by wa3lab