aredn/aredn #1118: Add black on green (lime green) screens (3270) for low light situations

kp4djt, 2024-03-25:
**Is your feature request related to a problem? Please describe.**
NO
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...
I use my screens in low light conditions many times. The best screen is a green text on black due to eye sensitivity being at about 555nm. I have tried the lime green screen which I have uploaded to some of the nodes. But every time there is an update I lose the screen and have to upload it again.
**Describe the solution you'd like**
A clear and concise description of what you want to happen.
I would like to see the green on black screen like the 3270 screens I used to see and work with. I can load the lime green 
screen into the node but as soon as it is updated it is lost.
**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
I manually upload the css file to a node as the only alternative. 
**Additional context**
Add any other context or screenshots about the feature request here.

aanon4, 2024-03-25:
If you have written this style I can add it to the build.

kp4djt, 2024-03-25:
OK I will get it to you. Here or on Slack?

aanon4, 2024-03-25:
Add it here - thanks.

kp4djt, 2024-03-25:
> Here it goes.

Thank you very much. That screen can be cranked way down in a dark EOC or
other windowless room
where the screens are kept dim.

kp4djt, 2024-03-25:
Here it goes.
***@***.***/home/kp4djt/Public/aredn/css/lime_on_black.css>

aanon4, 2024-03-25:
No attachment ... just a filename.

kp4djt, 2024-03-25:
That is weird, it shows in the sent mail.. But here goes again...

aanon4, 2024-03-25:
I've never seen someone successfully add an attachment to a github issue via email, and this does not seem to be an exception.

kp4djt, 2024-03-25:
Did you get the second one? if not I can send it some other way.

aanon4, 2024-03-25:
No. If you could log into github and attach via the website that would be best.

kp4djt, 2024-03-25:
[lime_on_black.txt](https://github.com/aredn/aredn/files/14749554/lime_on_black.txt)

I had to change the ext to txt to load it, GH did NOT like css.

closed 2024-03-27 by aanon4