aredn/aredn #1131: Nightly Build 20240328-f75e4d9 yields "Bad Gateway"

Orv, 2024-03-28:
After installing Nightly Build 20240328-f75e4d9 as a new install on a Rocket R5 AC Lite, attempts to save the configuration yielded Bad Gateway.  Attempts to reboot yielded the same error.  (Nodename was saved but IP addresses remained at the default of 192168.1.1)  After a power-cycle, the DHCP server was running

Support Data file attached.

[supportdata-NOCALL-202403230111.tar.gz](https://github.com/aredn/aredn/files/14795761/supportdata-NOCALL-202403230111.tar.gz)

aanon4, 2024-03-28:
Any chance you could make these changes on your node by hand and check it for me?

https://github.com/aredn/aredn/pull/1132

aanon4, 2024-03-29:
Please check the new nightly.

Orv, 2024-03-29:
Now works as expected.

closed 2024-03-29 by Orv