aredn/aredn #1150: CPE710 will not respond to ping after WIFI scan from connected node

wa3lab, 2024-04-03:
### Describe the bug
I have setup my test on channel 182, 10 mhz bandwidth, and LQM ON.  I did a FACTORY load of nightly build 20240327. Using WA3LAB-NanoM5 (Nano Station with 3.23.12.0 and direct connected laptop) as the connecting node.  Both the CPE710 (also with its own connected laptop) and the NanoM5 powered up and connected normally. I waited until both nodes LQM status was "active"(100% LQ/100 % NLQ).  I ran a WIFI scan from the NanoM5, and after a short pause the results screen came up with the normal expected response. I hit the quit button and returned to mesh status , which appears the normal 100% LQ/100 % NLQ. At this point if I click on the WV3P-CPE710 link from the Nano, it will time out in about 10 seconds indicating WV3P-CPE710 took too long to respond. At this point if I try to ping the CPE710 it will fail with no response. From the laptop plugged into the CPE710, the mesh status indicates 100% LQ/100 % NLQ, and all the remote nodes show up, and everything looks ok. If I click on the WA3LAB-NanoM5 link on the mesh status page, it will time out. Also pings from the CPE710 will timeout. If I go back to the NanoM5, I can successfully run a WIFI scan again, and it will complete normally, however the link and pings continue to time out. So, everything looks normal, but the nodes cannot ping each other. I have seen this same failure condition happen on powerup also. This condition will persist until I reboot the CPE710.  After some time the LQM quality status on the CPE710 will decrease below the minimum and the link will drop with" rf, blocked, latency".   
     Sometimes I can run the WIFI test OK once, but it will usually fail on the second time. Please understand I am just using the WIFI test to aggravate this situation to possibly help in debugging. 

- [ ] Does this issue still occur on the latest version? (**Required**)
- [ ] I tried a sysupgrade to 240401, and 240403 and both show 100% LQ and 0% NLQ on the CPE710 and LQM is "rf,idle".  The CPE710 does not show up on the NanoM5 at all.  Could not use these versions for testing.
- [ ] 
### Expected behavior
After the WIFI scan the node should be able to be pinged.

### Screenshots
Please include a screenshot of your system information (if possible) if the specific system environment is relevant to the bug.

If applicable, add screenshots to help explain your problem.

### Additional context
Debug data just after WIFI scan.
[supportdata-WA3LAB-NanoM5-202404031904.gz](https://github.com/aredn/aredn/files/14857598/supportdata-WA3LAB-NanoM5-202404031904.gz)
[supportdata-WV3P-CPE710-202404031905.tar.gz](https://github.com/aredn/aredn/files/14857604/supportdata-WV3P-CPE710-202404031905.tar.gz)

Debug data after CPE710 LQM status went to "rf, blocked, latency"
[supportdata-WA3LAB-NanoM5-202404031931.gz](https://github.com/aredn/aredn/files/14857613/supportdata-WA3LAB-NanoM5-202404031931.gz)
[supportdata-WV3P-CPE710-202404031930.tar.gz](https://github.com/aredn/aredn/files/14857615/supportdata-WV3P-CPE710-202404031930.tar.gz)

wa3lab, 2024-04-20:
Upgraded to 20240417 and this issue persists.

Orv, 2025-01-24:
Most current testing from Glenn:

    I retested the CPE710 with nightly build 1/25.  The node will fail with "unable to ping" on its own in anywhere from 1 to 12 hours.  In my 3 node test setup, I observed that the failure to ping would start with one node, while the other node could still be pinged OK.  Eventually, the second node would also become unreachable via ping.  While the CPE710 is unable to be pinged, the other 2 nodes in the test ping each other OK.  I also observed that running a WIFI scan on the CPE710 will tend to accelerate the onset of the failure.  This failure mode is essentially the same as open issue #1150 I submitted on April 3, 2024.

Glenn WA3LAB