aredn/aredn #1157: GL-AR300M

kc2cbd, 2024-04-07:
### What is the hardware/platform
GL.iNet GL-AR300M-ext

### Links
https://openwrt.org/toh/gl.inet/gl-ar300m

### Additional context
Why are we not building this anymore?
Thanks
Dale

aanon4, 2024-04-07:
Is this not it? http://downloads.arednmesh.org/afs/www/?version=20240405-0dba266&target=ath79%2Fgeneric&id=glinet_gl-ar300m16

kc2cbd, 2024-04-07:
No sir, The GL-AR300M and GL-AR300M-ext are 128mb nand devices

aanon4, 2024-04-07:
Dont suppose you have a link to buy the specific model? Nand devices have been finicky so I'll need to get one.

kc2cbd, 2024-04-07:
Here is one on ebay
https://www.ebay.com/itm/156148579621?itmmeta=01HTX3K7BTP9R88RWWPARQC8GR&hash=item245b2e3d25:g:8jEAAOSwN79mCxiZ&itmprp=enc%3AAQAJAAAA4FGBOk%2BKCl8yh%2FJzdfdD0ynAQWhswXzX3nvtouCvG4F%2Fh0gZ7ZR%2B7ZWGP%2FKxStfHiK69916auNvFmfTNH1MI2pwig5RtFcL9amXSjyED0v67cHy2%2FwSPA5WgkwFBk9pnEzQwHardCtlSRVz6Mvd0JVzozTQtZbc6WY5mwDhmIN6AqXF1zJtydorVccrdDVkTTZF0ieStHNxM9o3tYZGh1wj%2BkTexyTrdbj%2Fn7Rd2L40E%2FTIVOm8mHhJS%2B%2FlB%2FepmX2vQB573jfRSOkGhZFl0Xs%2F8AlCJUGIPgdACvmu57Xmu%7Ctkp%3ABk9SR4D2zKPXYw

kc2cbd, 2024-04-07:
GL.inet has a refurbished one as well.
https://store-us.gl-inet.com/products/refurbished-gl-ar300m-series-mini-smart-router?variant=47188207403299&currency=USD&utm_medium=product_sync&utm_source=google&utm_content=sag_organic&utm_campaign=sag_organic&srsltid=AfmBOoq3JopuwE7peoTKmCl8yoQQgSWkTlU307IOZR3J_87wOxMB5ONlys0

aanon4, 2024-04-07:
When I check the spec, for these the devices claim to be NOR flash. I do see a NAND flash version listed but can't find a link to buy one.

kc2cbd, 2024-04-08:
Per the device page (https://www.gl-inet.com/products/gl-ar300m/) the GL-AR300M is NAND and the GL-AR300M16 is NOR.

aanon4, 2024-04-08:
I've added it to the build here:
https://github.com/aredn/aredn/pull/1158
But its untested of course. Let me know if it works tomorrow once it's available.

aanon4, 2024-04-09:
http://downloads.arednmesh.org/afs/www/?version=20240409-0cb2ebe&target=ath79%2Fnand&id=glinet_gl-ar300m-nand

kc2cbd, 2024-04-09:
It seems to be working correctly.

closed 2024-04-09 by aanon4