aredn/aredn #1161: hAP AC2 losing Mesh RF when rebooted of tunnel changes are done

VA2XJM, 2024-04-16:
### Describe the bug
When updating the tunnel server configuration or rebooting a hAP AC2 on nightly `20240411-8627104`, the hAP AC2 RF 2Ghz radio drop all RF links and shows no heard signal. `LQM` is set to `OFF` and `FARTHEST Neighbor` is set to `1 km`.

The status page display `signal|noise|SNR:	no RF links` while there is 3 other nodes in range.

- [X] Does this issue still occur on the latest version? (**Required**)

### Expected behavior
Not having the RF mesh radio drop and work as before.

### Additional context
[supportdata-VE2ATU-FN18CC-Gateway-202404160959.tar.gz](https://github.com/aredn/aredn/files/14997987/supportdata-VE2ATU-FN18CC-Gateway-202404160959.tar.gz)
Support data from the local hAP AC2 node updated to latest nightly. Support data have been retrieved less than 3 minutes after updating tunnel settings and having the radio to drop while saving.

Right now checking if a watchdog is reactivating the RF radio past the 15minutes mark as requested by Tim KN6PLV.

VA2XJM, 2024-04-16:
After more than 20 minutes, I accessed the Wi-Fi scan page and hit the `Rescan` then the RF mesh is back alive.

I notice that the scan page shows nothing even few minutes after but is still connected:
[image]
[image]

aanon4, 2024-04-16:
The log file has pages of this:
```
[53820.102469] nf_conntrack: nf_conntrack: table full, dropping packet
[53821.507891] nf_conntrack: nf_conntrack: table full, dropping packet
[53821.554060] nf_conntrack: nf_conntrack: table full, dropping packet
[53821.554231] nf_conntrack: nf_conntrack: table full, dropping packet
[53824.789817] nf_conntrack: nf_conntrack: table full, dropping packet
[53826.272730] nf_conntrack: nf_conntrack: table full, dropping packet
[53832.389784] nf_conntrack: nf_conntrack: table full, dropping packet
[53832.389923] nf_conntrack: nf_conntrack: table full, dropping packet
[53833.239629] nf_conntrack: nf_conntrack: table full, dropping packet
[53837.394825] nf_conntrack: nf_conntrack: table full, dropping packet
[53837.394969] nf_conntrack: nf_conntrack: table full, dropping packet
[53845.474969] nf_conntrack: nf_conntrack: table full, dropping packet
[53851.131217] nf_conntrack: nf_conntrack: table full, dropping packet
[53853.634044] nf_conntrack: nf_conntrack: table full, dropping packet
[53855.177668] nf_conntrack: nf_conntrack: table full, dropping packet
[53860.475083] nf_conntrack: nf_conntrack: table full, dropping packet
[53860.682870] nf_conntrack: nf_conntrack: table full, dropping packet
[53861.351788] nf_conntrack: nf_conntrack: table full, dropping packet
[53862.782587] nf_conntrack: nf_conntrack: table full, dropping packet
[53862.807997] nf_conntrack: nf_conntrack: table full, dropping packet
[53864.648931] nf_conntrack: nf_conntrack: table full, dropping packet
[53864.922845] nf_conntrack: nf_conntrack: table full, dropping packet
[53872.111872] nf_conntrack: nf_conntrack: table full, dropping packet
[53872.126459] nf_conntrack: nf_conntrack: table full, dropping packet
[53881.082177] nf_conntrack: nf_conntrack: table full, dropping packet
[53882.087212] nf_conntrack: nf_conntrack: table full, dropping packet
[53882.979319] nf_conntrack: nf_conntrack: table full, dropping packet
[53884.103165] nf_conntrack: nf_conntrack: table full, dropping packet
[53886.340022] nf_conntrack: nf_conntrack: table full, dropping packet
[53887.982783] nf_conntrack: nf_conntrack: table full, dropping packet
[53888.841477] nf_conntrack: nf_conntrack: table full, dropping packet
[53899.572607] nf_conntrack: nf_conntrack: table full, dropping packet
[53901.505945] nf_conntrack: nf_conntrack: table full, dropping packet
[53905.785291] nf_conntrack: nf_conntrack: table full, dropping packet
[53905.793400] nf_conntrack: nf_conntrack: table full, dropping packet
[53906.970729] nf_conntrack: nf_conntrack: table full, dropping packet
[53907.692648] nf_conntrack: nf_conntrack: table full, dropping packet
[53917.665724] nf_conntrack: nf_conntrack: table full, dropping packet
[53918.357448] nf_conntrack: nf_conntrack: table full, dropping packet
[53919.094932] nf_conntrack: nf_conntrack: table full, dropping packet
[53930.885889] nf_conntrack: nf_conntrack: table full, dropping packet
[53932.286308] nf_conntrack: nf_conntrack: table full, dropping packet
[53942.792347] nf_conntrack: nf_conntrack: table full, dropping packet
[53943.424402] nf_conntrack: nf_conntrack: table full, dropping packet
[53943.525732] nf_conntrack: nf_conntrack: table full, dropping packet
[53943.529086] nf_conntrack: nf_conntrack: table full, dropping packet
[53943.903185] nf_conntrack: nf_conntrack: table full, dropping packet
[53943.972116] nf_conntrack: nf_conntrack: table full, dropping packet
[53944.191440] nf_conntrack: nf_conntrack: table full, dropping packet
[53945.164460] nf_conntrack: nf_conntrack: table full, dropping packet
[53948.429574] nf_conntrack: nf_conntrack: table full, dropping packet
[53950.165275] nf_conntrack: nf_conntrack: table full, dropping packet
[53952.112045] nf_conntrack: nf_conntrack: table full, dropping packet
[53955.012755] nf_conntrack: nf_conntrack: table full, dropping packet
[53955.434917] nf_conntrack: nf_conntrack: table full, dropping packet
[53971.494304] nf_conntrack: nf_conntrack: table full, dropping packet
```
Is this node anything special? The conntrack table is quite large so this is odd and suggests a massive number of connections running through this node.

VA2XJM, 2024-04-16:
This is node is one of the two node that are our border nodes and both are hAP AC2 (the 2nd one being the backup and remote). We have our main server hosted behind this specific node. This server host most of our services and many are available via VPN for our members. This may drive some traffic, but the average traffic for the last 12 hours is bellow 5mbps.

We are redesigning the network, thus why the update for WireGuard so regional tunnel (used as backup to RF) are migrated on WG and we prepare to use a second device to act as supernodes for linking with networks out of our operating area.

Prior to update everything was fine with 3.23.4.0.

VA2XJM, 2024-04-17:
Ok, so I changed the value for nf_conntrack : `echo 48256 > /proc/sys/net/netfilter/nf_conntrack_max` and then `cat /proc/sys/net/netfilter/nf_conntrack_count` return averaging between 28 000 and 34 000 without impacting the node load or memory use. It eliminates the complains. It do not seems to change performance. I may make put the value permantly in `/etc/sysctl.d/10-default.conf`

On the RF side, I notice that the node goes back on RF once in a while for few minutes until it drops again. Last time I saw that at another location, the hAP was inside a cabinet and the TX power needed to be lowered in order to stop what I believe to be signal reflexion that could  have caused some "echo" that makes the node goes deaf. Since then, I always lower TX power on my hAPs and that eliminates those problems. 

This specific hAP is set to 18dB in AREDN. However iwconfig seems to indicate that it is 30dB and the AREDN max is 22dB. I could be wrong on that, but maybe a clue.
```
root@VE2ATU-FN18CC-Gateway:~# iwinfo
wlan0     ESSID: "AREDN-10-v3"
          Access Point: 18:FD:74:7A:B9:5A
          Mode: Ad-Hoc  Channel: -2 (2.397 GHz)  HT Mode: HT20
          Center Channel 1: -2 2: -1
          Tx-Power: 30 dBm  Link Quality: 53/70
          Signal: -57 dBm  Noise: -95 dBm
          Bit Rate: 11.0 MBit/s
          Encryption: none
          Type: nl80211  HW Mode(s): 802.11b/g/n
          Hardware: embedded [Qualcomm Atheros IPQ4019]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy0

```

aanon4, 2024-04-17:
iwconfig says 30 by default - it's not a trustworthy value.

VA2XJM, 2024-04-17:
So back to the Wi-Fi scan returning only the affected device (https://github.com/aredn/aredn/issues/1161#issuecomment-2059213116)

Set TX Power lower:
```
root@VE2ATU-FN18CC-Gateway:~# iw wlan0 set txpower fixed 1600
root@VE2ATU-FN18CC-Gateway:~# iwinfo
wlan0     ESSID: "AREDN-10-v3"
          Access Point: 18:FD:74:7A:B9:5A
          Mode: Ad-Hoc  Channel: -2 (2.397 GHz)  HT Mode: HT20
          Center Channel 1: -2 2: -1
          Tx-Power: 16 dBm  Link Quality: 55/70
          Signal: -55 dBm  Noise: -94 dBm
          Bit Rate: 29.5 MBit/s
          Encryption: none
          Type: nl80211  HW Mode(s): 802.11b/g/n
          Hardware: embedded [Qualcomm Atheros IPQ4019]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy0
```
Then nearby devices appears into the scan:
[image]

I will check if the link holds on during the night but so far it is back and holding longer than it was since update.