aredn/aredn #1162: Bullet M5 - Bad Gateway

VA2XJM, 2024-04-16:
### Describe the bug
Bullet M5 return bad gateway when trying to access the setup page. However other pages can be accessed through direct URL.

Current Version: 20240411-8627104      Hardware Type: (ath79/tiny) ubnt (bullet-m-ar7241)

- [X] Does this issue still occur on the latest version? (**Required**)

### Expected behavior
Access the setup page without error.

### Screenshot
[image]

[supportdata-VA2XJM-BulletM5-202404161123.tar.gz](https://github.com/aredn/aredn/files/14999261/supportdata-VA2XJM-BulletM5-202404161123.tar.gz)

aanon4, 2024-04-16:
Unfortunately the log file which I really need to debug this is empty. Can you reboot, generate the Bad Gateway, and then dump the support data with as few other steps as possible? Probably the logger crashed.

VA2XJM, 2024-04-16:
Reboot, Login + error, Support Data.

Please note the version to latest stable (I am downgrading the unit to see if a working version is available.

[supportdata-VA2XJM-BulletM5-202404161152.gz](https://github.com/aredn/aredn/files/14999450/supportdata-VA2XJM-BulletM5-202404161152.gz)

VA2XJM, 2024-04-16:
It fixed the thing for the Bullet. Once fixed, I made an update on a GL.Inet ARM300 and faced the same problem and the solution also fixed the problem.

It would be interesting to have a check on the value and a coded  "reset" to restore proper values when meeting null or negative value.

I keep that in my notes. I'll let you close the ticket if you wish to keep it for dev purpose or not.

Thank you for this rapid response.

aanon4, 2024-04-16:
Unless there's something you can tell me about what those nodes were running and how they were configured before the update, then I'lll close this. If you have others and want to do a data support dump before the upgrade just in case they have the same problem that would be good to have.

aanon4, 2024-04-16:
Reopen if we get some before and after data support info.

VA2XJM, 2024-04-16:
Here you go, with a Bullet M2. Pre-update, first leg of the update (to 3.23.4.0), hit the error + FIX and then latest nightly.
[supportdata-VA2XJM-BulletM2-202404161643.gz](https://github.com/aredn/aredn/files/15000327/supportdata-VA2XJM-BulletM2-202404161643.gz)
[supportdata-VA2XJM-BulletM2-202404161643.gz](https://github.com/aredn/aredn/files/15000499/supportdata-VA2XJM-BulletM2-202404161643.gz)
[supportdata-VA2XJM-BulletM2-202404161639.tgz.gz](https://github.com/aredn/aredn/files/15000501/supportdata-VA2XJM-BulletM2-202404161639.tgz.gz)
[supportdata-VE2ATU-Operation-BulletM2-202404161347.tar.gz](https://github.com/aredn/aredn/files/15000824/supportdata-VE2ATU-Operation-BulletM2-202404161347.tar.gz)

The M2 is the same setup as the M5 so node with AP activated in NAT mode.

aanon4, 2024-04-16:
Nat mode .. that seems important .. as almost no one uses it. Let me see.

VA2XJM, 2024-04-16:
Since those 2 bullets are used as Wi-Fi AP for mobile devices, I went with NAT mode for the practical side of not having to think about how many devices are connected and bust the maximum number of IP available.

aanon4, 2024-04-16:
https://github.com/aredn/aredn/pull/1163

aanon4, 2024-04-16:
If you have a way to test this fix tomorrow once's it in the build, that'd be lovely.

VA2XJM, 2024-04-16:
Just need to know the time of the nightly build and I'll see to make tests when I am back from work.

aanon4, 2024-04-16:
The nightly build completes about 1pm UTC. The next one with start with the build number 20240417-

VA2XJM, 2024-04-19:
Sorry for late reply. I made the tests tonight and it now works properly. Flashed the latest release, configured as NAT and then upgrade toward latest nightly. All good so far, will do the other unit and the ARM300 when off. Closing as fixed.

closed 2024-04-19 by VA2XJM